prusa3d/Prusa-Firmware-Buddy #4638: [BUG] Invalid QR code for print fan error page

ygoe, 2025-04-29:
### Printer model

Mini

### Firmware version

6.2.4

### Printing from...

Original USB drive

### Describe the bug

I have an error shortly after the second layer started. The print pauses and the display shows this:

[image]

My phone browser just redirected me to the Prusa home page. Two times. So the code/URL is unusable.

The error is probably also invalid. The fan is off in layer 1 and at 17% in layer 2, the display gives me around 600 rpm and I can see the fan spinning. With higher layers, the % and rpm increase and after a while, at 50% (max by print setting), I can feel (and smell) the fan blowing air at my face when I'm sitting in front of the printer. So on the hardware side, everything is fine.

I had to disable the fan check to keep printing. I never had this before and was waiting for the result elsewhere. When I came back in between, I found everything in silence, the first layer printed and couldn't resume in that state because some filament had already gathered below the nozzle. On my second try, the same error occured, but I could quickly confirm it and resume.

Sithrazer, 2025-04-29:
QR codes for Marlin Timeout (prusa.io/31530) and Loadcell Timeout (prusa.io/31528) on the Core One firmware 6.3.2+10364 also redirect to the Prusa3d home page.

JakoobCZ, 2025-05-30:
Hi, @ygoe @Sithrazer thanks for reporting the URLs not working, we will fix them.

closed 2025-08-07 by github-actions[bot]

pecekd, 2025-10-06:
The bug was fixed.