prusa3d/Prusa-Firmware-Buddy #4640: [FEATURE REQUEST] Overwrite printing temperature before printing

seppelicous, 2025-05-02:
### Printer model

MK4s, Mk4, XL

### Describe the feature

During slicing you choose your filament type, like PLA. But filaments are different and maybe I want to print the same model with different PLA brands. While Prusament PLA is doing fine at 220°C, you might have a lot of stringing with others like eSUN or Sunlu. Here 205°C is the best temperaure, while other settings remain the same.
Now I have to slice the same model with different temperatures and have to add the temperature information into the filename to select the right file.

**Channing the nozzle temperature during print is not an option!** To do so, you have to wait at least for the second layer, as the layer change 1 to 2 overwrites any manual adjustment.

### Expected functionality

I would like to see an option to overwrite the nozzle temperature in the print file.
Having a seperate step after starting the print like "Choose PLA nozzle and bed temperature" would be really helpful!
Advanced: Use user defined presets like "PLA eSUN", "PLA+" as a selection with modified nozzle/bed temperatures.

murk-sy, 2025-05-04:
Similar issue/possibly duplicate:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3343

Prusa-Support, 2025-08-16:
Sometimes, the nozzle temperature is not the only difference between filament profiles, even for certain filaments of the same material, so reslicing should probably remain the best practice.

Other than this, we may agree on the fact that this feature request is actually already covered by issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3343.

Michele Moramarco
Prusa Resasearch

closed 2025-08-16 by Prusa-Support