prusa3d/Prusa-Firmware-Buddy #4642: [BUG] Manual color change spits out a cm of additional filament after the purge.

CpuNut, 2025-05-06:
### Printer model

MK4, MK4S, and I assume others.

### Firmware version

All

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

On the MK4 and MK4S on a manual color change, after the purge and pressing "Yes" it spits out about a cm of additional filament which I always have to catch with pliers so it won't mess up the print.  I was told this is intentional but can't imagine why.

If someone can point me to the source file and line(s) that I could change or comment out to stop this it would be appreciated :-)

### Expected behavior

On a manual color change, after the purge and selecting Yes, I would like it to not purge the additional cm of filament before the layer with the new color starts.

dunkmann00, 2025-05-11:
I thought this behavior was in the slicer? In the `Custom G-code` section of the printer, the `Color Change G-code` I see set (for Prusaslicer) is:
```
M600
G1 E0.3 F1500 ; prime after color change
```

That second line is the one doing the extra extruding. Perhaps try removing that and see if it fixes your problem?

CpuNut, 2025-05-11:
Great thanks for the feedback!  Maybe this is why I wasn't finding it in the printer source code hehe 🙂

maru-sama, 2025-05-14:
Would it be possible to do the initial prime on the front right side instead, of course you have to remove the previous prime line first.

dunkmann00, 2025-05-14:
The "prime" after a color change is not like the prime that happens at the start of the print. This one just extrudes out some filament into the air, not onto the bed.

Your question is probably better suited for a separate issue. And if you do open a new issue, it would be better to do so in the [Prusaslicer](https://github.com/prusa3d/PrusaSlicer) repo, not here.

3d-gussner, 2025-08-11:
When you initiate a `M600` or color change it will indeed purge filament.
We had this discussion on the MK3S repository see [here](https://github.com/prusa3d/Prusa-Firmware/issues/2502) 

The purge of the filament is used to build up pressure after the change in the nozzle to prevent under extrusions.
So if we disable that extra extrusion you may continue with under extrusions, which is especially annoying on perimeters.

CpuNut, 2025-08-11:
> Correct me if I'm wrong (I can't check it now), but it seems that Core One doesn't do this (when filament purge is complete, and you press yes to continue printing, it doesn't purge additional amount, unlike MK4). If that is the case, then there is a solution for this inconvenience already implemented in Core One.

It also does it on the Core One.  I can see that it helps with under extrusion after a color change but it purges way too much and messes up the print, for example a QR code switching from white to black and in this case is not good.
I tried everything I could in Slicer in the color change GCode section, and it makes no difference.  I'd like to change it in the printer source code and set it to less purge on my printers.
Can someone point me to the file and label where I can make a change?  I've tried a couple things and only succeeded in killing all purges.  It seems to me there must be something unique to M600 in the code.  I’ve searched on everything I can think of in the code and didn't find anything unique to just M600 that only changes the purge in this one instance.
Thanks, Craig

dunkmann00, 2025-08-11:
> I tried everything I could in Slicer in the color change GCode section

Just curious what are some of the things you tried?

closed 2026-05-28 by github-actions[bot]