prusa3d/Prusa-Firmware-Buddy #4645: [BUG] Crash (BSOD) while loading custom filament

5jvm0u4, 2025-05-08:
### Printer model

MK4S

### Firmware version

6.2.4+8909

### Upgrades and modifications

None

### Printing from...

Not printing

### Describe the bug

Filament>Load Filament>PETG-CF(custom)
and it crashed.

### How to reproduce

Not consistent.

### Expected behavior

Expected to not crash.

5jvm0u4, 2025-05-08:
PR32018597378

coasterboy101, 2025-05-10:
I would like to confirm this as well, seems to only occur once, then it works normally, not sure what the exact trigger for it to happen again is. I can provide dump files if that helps, also, I have an MMU3 attached, dunno if that changes anything.

5jvm0u4, 2025-05-10:
I also have a MMU3 attached but disabled.

3d-gussner, 2025-07-10:
I have been trying to reproduce the issue but couldn't.

Here my steps I tried:
- MMU3 was enabled before the power off
- Power on printer
- MMU3 connected
- Disabled the MMU3 via LCD Settings
- Load filament ✔️
- Unload filament ✔️

- Enabled MMU3, still being powered on from previous test
- Load filament 2 ✔️ 
- Unload filament ✔️ 

- Disabled MMU3
- Power off and on
- Load filament ✔️ 
- Unload filament ✔️ 

Did you do something else? Can you try to describe the steps? @5jvm0u4 @coasterboy101 

@5jvm0u4 What is your PETG-CF custom looks like? Which temps do you have?
In PrusaSlicer profile for PETG-CF Prusament:
- Temp is 270°C
- Idle is 130°C
- Abrasive material

coasterboy101, 2025-07-10:
I can't speak for other users, but I didn't do anything special. The profile is just a copy of the polycarbonate profile with the Abrasive Material option toggled on. Noteably, it seems to only happen once per reboot, after the printer reboots once, the profile works fine.

3d-gussner, 2025-07-10:
@coasterboy101 But does it crash "once" each time you turn on the printer?

coasterboy101, 2025-07-10:
Yes it does. To be clear, you have to choose the "Load Filament to Nozzle" option. Not just load the filament to the mmu.

5jvm0u4, 2025-07-10:
Late for the conversation😅
For me, the crash usually happens after having several prints following up by long idle time(for hours~days) when it's on but not in use, and then when I click on the custom filament it will freeze for about a second before BSOD shows up, once reboot it won't crash anymore.
It happens to both of my MK4S, both having MMU3 connected, more than 5 times combined.

[image]

5jvm0u4, 2025-07-14:
Good news, it's consistent, trying to narrow down tha exact procedure to reproduce this crash, will report back soon.

[image]

5jvm0u4, 2025-07-14:
Hardened nozzle isn't the cause, turned it off and it still crashed.