prusa3d/Prusa-Firmware-Buddy #4649: [FEATURE REQUEST] Use higher temperature of previous and new filament at change

CNCKitchen, 2025-05-14:
### Printer model

CORE One, MK4, XL

### Describe the feature

I regularly switch between different filaments with varying melt points. Each time I switch from PCBlend back to PLA, the nozzle clogs or doesn't purge properly because the remaining PCBlend in the nozzle doesn't melt at 215 °C.

The printer is aware of both the previous and new materials, so why not utilize the higher of the two nozzle temperatures during the change to ensure that I can adequately purge all the leftover material? You could limit it to a specific value to prevent unwanted degradation. For example, Bambu's machines always switch filaments at 250 °C, which is quite reliable.

albjerre, 2025-05-15:
Yes, we need this! I had the same issues with nozzle clogs on Prusa XL, before I made a workaround.

On my Prusa XL, when changing from a high temp filament to a low temp filament, I currently have to use a workaround where I select the high temp filament as the new filament. Then I change to the filament, purge three times and then initiate a new filament change where I select the low temp filament as the new filament. This is cumbersome and could easily be avoided if the firmware always used the temperature for the high temp filament when changing.

andersnygaard, 2025-05-16:
Yes, the problem is prominent when changing from PETG to PLA and it only heats the nozzle to PLA temperature. I had a case where the remnants of PETG left in the nozzle just managed to purge during filament change and only after starting a new print, the rest of the PETG would suddenly release, ruining the print on layer four or five.

Tupson444, 2025-05-17:
This has been suggested in #1821. Counter argument says that changing from e.g. PC to PLA at PC temp can result in PLA carbonizing, which is a valid point, but the current solution to purge at lower temperature is even worse.

It should either be set to simply purge at higher temperature, or implement a good solution (e.g. https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1821#issuecomment-2506887494), not the way it is now.

CZDanol, 2025-05-29:
Internal ticket: BFW-7089

kyngs, 2026-09-24:
Still an issue