prusa3d/Prusa-Firmware-Buddy #4652: [ENHANCEMENT] Changing from high temp filament to low temp filament can clog nozzle

tomhammond, 2025-05-15:
### Printer model

XL Multi

### Describe the enhancement

When changing from a high-temp filament to a low-temp filament via the "Change filament" command on the printer's control panel, there is a danger of clogging the nozzle. This is what happens:

With ABS filament already loaded, I choose the "Change filament" command.

1. The printer heats to ABS temp, then unloads the filament.
2. The printer prompts for the new filament type, I choose PLA.
3. At this point, the printer waits for me to insert the PLA filament.
4. HOWEVER, at this point, the nozzle is cooling to PLA temp. ABS residue is still in the nozzle which may not clear out at PLA temp.

Here is what i have to do to prevent nozzle clogs when going from ABS to PLA:

1. With ABS filament already loaded, I choose the "Change filament" command.
2. The printer heats to ABS temp, then unloads the filament.
3. The printer prompts for the new filament type, I choose ABS (even though I am going to use PLA.
4. At this point, the printer waits for me to insert the PLA filament. I insert the PLA filament at it loads it as ABS, fully clearing out the ABS residue from the nozzle as the PLA passes through.
5. I choose "Change filament" again, the printer unloads it.
6. The printer prompts for the new filament type, I choose PLA and finish loading the filament.

It would be better to set the nozzle to the filament temperature of the EXISTING filament if the user selects a LOWER temperature filament to change to. This ensure that the existing hig-temp filament is flushed fully from the nozzle.

Tupson444, 2025-05-17:
Already suggested in #1821 and recently in #4649. This is a real issue and it should be fixed.

Prusa-Support, 2025-05-27:
Thanks for spotting this duplicate issue.
Our developers and product managers are still making considerations about pros and cons of automatic temperature picking.
I kindly suggest closing this issue in favour of #1821, and please feel free to take part in the existing discussion.

Michele Moramarco
Prusa Research

closed 2025-08-04 by github-actions[bot]