prusa3d/Prusa-Firmware-Buddy #4653: Cold priter fails to unload filament, firmware crashes into endless loop

pstimpel, 2025-05-16:
### Printer model

Core One

### Firmware version

6.3.2

### Upgrades and modifications

Stock printer, built in factory

### Printing from...

none

### Describe the bug

Printer was cold. Switched it on, and attempted to remove filament, ASA loaded. So "unload filament", pribter starts two things at the same time:

1.) heating the nozzle
2.) some kind of homing

The homing goes on and on without coming to and end. After 2 minutes it crashes. After reboot, the machine tells that belt tention is wrong. You can start over the whole thing by just telling the printer to unload filament again. 

On a sidenote: I find it dangerous, that the nozzle fan is offline while the red screen is showing an wait for confirmation. The nozzle was at 200° at that time.

ThomasHalwax, 2025-05-26:
https://github.com/user-attachments/assets/10c7c661-b1ae-40b4-b095-d1da2c903f0d

Same problem here with exactly the same environment (Core One factory assembled, same firmware)

pstimpel, 2025-08-04:
Wow, not even an answer ...

3d-gussner, 2025-08-06:
Sorry for the late answer. I assigned myself to this ticket so stale bot doesn't close it.

pstimpel, 2025-08-15:
Ok, here is the latest finding by me:

6.3.2: it happens regularly

6.3.4: the frequency of the crash dropped significant. I was able to reproduce once, but it worked most of the time. About 10 attempts in total. Also this crazy homing is not there anymore.

3d-gussner, 2025-08-15:
@pstimpel Thanks for the answer and feedback.

3d-gussner, 2025-08-20:
@ThomasHalwax Thanks for the feedback, very appreciated. Glad to see that it is solved for you.