prusa3d/Prusa-Firmware-Buddy #4655: [FEATURE REQUEST] Service Interval Reminder

koombot, 2025-05-18:
### Printer model

Core One, MK4, MK4s, XL, Mini

### Describe the feature

Car service intervals are based on the distance travelled or time since last service.  My car can track the number of miles I have driven and will remind whenever I start the car that  service is due in x miles.

My 3D Printer should also tell me when a service is due.

### Expected functionality

There should be default service interval period for each printer (more accurate that every couple hundred hours).   The user should be able to change the interval period.  
As the printer approaches this interval period it should have a pop up on the printer and in PrusaLink/Connect that a service will be required in x hours of print time, or if the current print will take it over the interval that a service will be required after the print.  

- These warning should not prevent a print from starting and the user should have the ability to either disable the popups permanently (ie prevent them from ever appearing) or temporarily (they have started appearing but the user wishes to ignore).  
- The user should be able to also say "Service Completed" to reset the timer.  
- If the popup is integrated on the Connect/Link apps it should provide a link to a guide on what the user should do to service the machine.
- As a stretch goal the feature could also be 'filament weighted' where the time printing has a weighting factor based on the filament used where printing more hours with higher temperature filaments will reduce the interval time compared to lower temperature filaments.

This feature would integrate well for new users who may be unaware of the requirement to service the printer as well as users who are running the printer in a more commercial environment by allowing them to ensure services are tracked and can be planned for.

albjerre, 2025-05-24:
I made a similar request a while ago:
[https://github.com/issues/created?issue=prusa3d%7CPrusa-Firmware-Buddy%7C4526)](https://github.com/issues/created?issue=prusa3d%7CPrusa-Firmware-Buddy%7C4526)

3d-gussner, 2025-07-24:
@koombot Thanks for the suggestion, as @albjerre already pointed out a similar request has been filed. I will close this ticket. If you have any input please leave a comment in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4526

closed 2025-07-24 by 3d-gussner