prusa3d/Prusa-Firmware-Buddy #4661: Side Filament sensor delete/disable

csweeney0926, 2025-05-22:
### Printer model

XL 2 tool

### Describe the feature

The side filament sensors have proven to be highly unreliable (for me) on my Prusa XL. 

The tool 1 sensor got jammed in the open position early on, was cleaned and restored to functionality, and then jammed again in the same fashion. This was a relatively minor inconvenience as the extruder filament sensor works just fine and all I do is pull the Bowden tube to remove the end at each filament roll run-out.  

My tool #2 side filament sensor also failed but was more problematic as it failed and got stuck in the closed position (so filament cannot be fed into the sensor at all). My side sensor is not mounted on the printer but rather on a dry box enclosure that makes the sensor not easy to access for maintenance and so I simply removed the Bowden tube from the sensor from tool 2 and feed the filament to tool 2 directly, bypassing the filament sensor entirely. 

### Expected functionality

I looked at the settings and there is an option to disable the filament sensors but it appears to disable all the sensors (extruder and side mount). I would like a feature to be added to disable the side mount sensors while leaving the extruder sensors functional. This would allow me to remove the side mount filament sensors entirely as they have only made the printer less reliable.

rcmaniac25, 2025-05-27:
Settings -> Filament Sensors -> can turn on/off the side sensors while leaving the extruder sensors

csweeney0926, 2025-05-27:
> Settings -> Filament Sensors -> can turn on/off the side sensors while leaving the extruder sensors

Really? Last time I looked there was only the option to turn off all filament sensors. Maybe I need to update my firmware if this feature already exists (or maybe it needs to be more clearly labeled in the menu). I'll have to update and look for this function.

closed 2025-08-04 by github-actions[bot]