prusa3d/Prusa-Firmware-Buddy #4662: [BUG] Prusa XL Crashes If Stopped Mid Print & Then Tool Parked that is Not Tool 1

petgmelter, 2025-05-23:
### Printer model

XL 5 Head

### Firmware version

6.2.5+8912 (latest available as of April 28, 2025)

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

Steps to recreate the bug:

-     Start a print job with PETG loaded in tool #1 and tool #5.
-     Stop the print job in mid-print when tool #5 is actively being used
-     At this point the printer moves tool #5 to the rear left, in front of tool #1.
-     On the printer Control screen, instruct the printer to park the current tool.
-     At this point the printer starts a homing sequence and one of the first things it does is move the Y axis towards the rear of the printer.  Since tool #5 is still in the tool changer, tool #5 crashes into tool #1, which is parked.

This is reproducible 100% of the time.

Also note, one time when tool #5 crashed into tool #1, the printer operating system completely crashed and my printer lost its entire configuration, forcing me to go through the entire printer re-calibration (~1 hour).

The suggested fix is, if a print is stopped mid-print, either don't move the print head so close to tool #1, or park the current tool before moving the tool changer in front of tool #1.  If the tool changer doesn't have a tool attached to it, then it won't crash into tool #1.

hagre, 2025-05-24:
[BUG] head can crash after reassignment #4588

petgmelter, 2025-05-24:
Thanks hagre.  Bug 4588 does indeed have the same outcome.  It's a pain cause if it collides with tool 1 too severely, it can knock tool 1 out of alignment.

morphias2004, 2025-05-29:
Happens with any tool other than Tool 1 (0).

Also often makes the next print you start have a head crash at the start or just rams the head into the front left and makes the stepper motors scream. Been there for many firmware revisions, including the latest.

morphias2004, 2025-07-21:
The XL just isn't getting the attention since the CORE One was released. :(

petgmelter, 2025-09-09:
Prusa support asked me to make a video demonstrating the problem.  I will not do that because I don't want to damage the machine and don't want to waste another hour re-calibrating everything after the crash happens.  Plus, it has been easily reproduced 100% of the time by many users.  If the Prusa firmware developers would just take some time to even try and reproduce the problem themselves, I'm convinced it could be easily fixed.

eqdeng, 2025-09-10:
> Prusa support asked me to make a video demonstrating the problem. 

@petgmelter If I get a chance I'll make a video. Are we supposed to post it here, or send it to them directly?

crasbe, 2026-01-16:
Issue #5009 seems to describe the same phenomenon.

CZDanol, 2026-01-19:
Okay, after some investigation:
* #5009 seems to be a different issue
* I have not been able to reproduce this one on 6.4.0 final.

Could you please check whether the crashing is happening with this g-code on the 6.4.0-final?
```
G28 ; Home

T1 ; Pick tool 2

G0 X100 Y100 F10000 ; Move the head somewhere

; Wait, give the user an opportunity to abort the print
G4 S5
G4 S5
G4 S5

; filament used [mm] = 5177.58, 5697.73, 0.00, 0.00, 0.00
; filament used [cm3] = 12.45, 13.70, 0.00, 0.00, 0.00
; filament used [g] = 15.44, 16.99, 0.00, 0.00, 0.00
; filament_type = PLA;PLA;PLA;PLA;PLA
```

If no, can you provide a gcode where it is happening?

closed 2026-09-14 by github-actions[bot]