prusa3d/Prusa-Firmware-Buddy #4672: [BUG] C1 BuddyCam no Video feed

Chips83, 2025-05-28:
### Printer model

Core One

### Firmware version

6.3.3

### Upgrades and modifications

None

### Printing from...

USB Drive 

### Describe the bug

Installed the C1 BuddyCam and there is no Video Feed. There is only a snapshot / photo every 10 seconds.
Nothing mentioned at the Shop Website or in the manual so it is a bug for me.  

Also it is only black and white, but it is mentioned at the manual that this will be a future firmware update. 

So I don’t like to buy products where features and functionality is maybe coming soon, because sometimes it will never find the way into the product. 

But I had no chance to know that because there is no Information at the shop website. It’s like a deceiving the customer. 

Who knows if it will come or when it will come?!?

tbedore, 2025-05-30:
I think that’s a prusa connect limitation actually. Number one complaint about the buddy camera is that it only does black and white snapshots and not color video.

zuidwijk, 2025-06-10:
Please read the bottom of the [product page](https://www.prusa3d.com/product/buddy3d-camera-for-prusa-core-one/) :

**Q2**
- Basic camera controls (Status, rotation, Day/Night mode, volume settings, etc.)
- On-demand snapshot
- Updating camera firmware using the Prusa mobile app
- Time-lapse support (storing images on the SD card)
- Video streaming over local network

**Q3**
- Camera controls (Light on/off, flash, resolution change)
- Camera name settings
- Remote camera restart

**Q4**
- Video streaming over internet
- SFTP for remote content download