prusa3d/Prusa-Firmware-Buddy #4677: [BUG] Core One - Make automatic loading of new gcode-files configurable as it leads to unwanted file-locks

zeus86, 2025-06-02:
### Printer model

Core One

### Firmware version

6.3.2

### Printing from...

PrusaLink

### Describe the bug

When sending a new gcode-file over PrusaLink while the Printer is otherwise idle, the printer loads said g-code file so that the user just have to press "print" on the printer to continue. However, I'd like to see this "feature" configurable, as it leads to very undesireable file-locking. As long the Printer is waiting for confirmation, you cannot delete the file (or any folder above as a consequence of that). 

I do run quite often into this problem, where I slice some files, and before printing changing something and reupload-files. I cannot use the same filename, or rename folders as long the printer is locking the file waiting for confirmation. I'd assume, for Users, that immediately want to print their files right away, there is the "upload & print" option available, for everyone else it would be desireable to at least alter the default behaviour and suppress automatic loading of uploaded files. When the Printer is in another room, this is much more annyoing than it sounds like. 

### How to reproduce

* Have printer beeing idle
* Upload g-code-file via PrusaLink
* Try to delete/rename/move the uploaded file
  * this results in an error (http/500) because the printer is locking the g-code file while waiting for confirmation, making alterations to the uploaded file (or any of it's parent folders) impossible unless dismissed physically at the printer itself 

### Expected behavior

* Autoloading beeing configurable
* better error-messages

murk-sy, 2025-06-09:
+1

Automatic file loading should *only* happen after USB drive plug in. I use "Upload only" a lot since more often than not I have to load different filaments before printing on XL. If you use "Upload only", that would mean you don't intend to print immediately.

Plus, when printing on Minis, sometimes (haven't figured out why yet) after a print is finished (using "Print immediately"), after closing the print finished screen to go to the main menu, I sometimes get prompted to print the file that has just been completed.

Having some sort of a tray icon to indicate that a file has been uploaded (dismissed by opening the print pick screen) would be nice though.

zeus86, 2025-08-11:
This is still an open Problem, just noone has taken care of yet...

3d-gussner, 2025-11-17:
> [@3d-gussner](https://github.com/3d-gussner) are there any updates on this in the internal ticket?

Sorry but at this moment there is no update in the internal ticket.

zeus86, 2026-09-28:
This is still an open Problem.