prusa3d/Prusa-Firmware-Buddy #4678: [BUG] Core ONE upgrade consistantly fails MBL probing sequence in back right corner

discojon, 2025-06-05:
### Printer model

Core ONE

### Firmware version

6.3.3+10442

### Upgrades and modifications

MMU

### Printing from...

Prusa Connect

### Describe the bug

If a print requires the back right (150,100 to 240,200) to be probed, the printer will consistently fail bed leveling.  If the model is relocated in the slicer so that area doesn't need probed, the probing sequence will complete without issue.  This issue can be reliably eliminated by stacking two build plates (reinforcing / minimizing deflection).  

### How to reproduce

Place a model in the indicated area of the build plate, slice, and run gcode.  Disabling heaters during probing has no effect.  The only reliable fix I've found is to double up build plates (Granted I don't allow the print to begin since there is no stability in the top plate).  A gcode that will reliably fail with one build plate (specifically the MK3 Powercoated build plate), will complete probing with stacked plates.

### Expected behavior

Printer will probe, bed level, and begin printing

discojon, 2025-06-05:
Update:  Lowering the nozzle ~1mm also allows the probing sequence to complete without issue.

colinb88, 2025-06-13:
Just finished conversion MK4S to CORE One and have the same problem. Using two build plates works, suggesting the problem is related to mechanical flex and possibly related to the single lead screw at the back. I suspect the touch sensor is not sensitive enough and with the bed flexing, the sensor is not triggering.

3d-gussner, 2025-10-28:
@colinb88 Great to hear you could solve the issue. Can you please send a picture?