prusa3d/Prusa-Firmware-Buddy #4679: [BUG] Toolhead crashes into right side dropping tool multiple times in a single print

murk-sy, 2025-06-06:
### Printer model

XL 5 tool

### Firmware version

6.2.5+8912

### Upgrades and modifications

None

### Printing from...

USB/Connect

### Describe the bug

When printing a specific file, on 3 specific occasions in one specific object, the toolhead rams into the side of the printer, causing the tool to drop.
This has happened on 2 consecutive prints.

2 videos:
https://youtu.be/WgnXdrmQBsA
https://youtu.be/33L2I-fOaIY

### How to reproduce

Print the entire gcode file. Extracting just the problematic layers does not reproduce the issue on my machine.

Loaded filaments:
- Tool 1: PETG (unused)
- Tool 2: PETG (unused)
- Tool 3: empty
- Tool 4: PETG supports (tool 4 in slicer)
- Tool 5: PLA (mapped from tool 1 in slicer)

- 1st crash: After 9h12m somewhere before Z 149 mm
- 2nd crash: After 9h 18m, just before or during Z 149.2 mm
- 3rd crash: After 9h 23m, just before or during Z 150.0 mm

This may be related to error accumulation bug.

Note: After the print completed earlier today, I ran trimmed gcode hoping to reproduce the issue, but it ran perfectly through it, so I don't believe it's gcode related. **However, when starting a new print, the head moved diagonally at the beginning of XY homing.**

### Expected behavior

- Prevent excessive X travel

### Files

The following has been sent to reports@prusa3D.com
- bgcode
- memory dump
- 3mf file

Prusa-Support, 2025-08-06:
Hi @murk-sy, thanks for opening the issue and sharing your files.

Please mind including the reference code that comes via automatic email reply after every email sent to reports@prusa3D.com.
I think I have found your email, and your reference code is: **PR32274532738**.

Our developers showed interest in getting these files, but they apparently were shared via a link, that probably expired.
Please consider sending the files again, ideally not compressed and directly attached to the email (the listed files should small enough).
I've sent an email back to you about this. Please attach the files in reply to my email if you don't mind. 🙏 

Michele Moramarco
Prusa Research

murk-sy, 2025-08-06:
Replied to your email with the updated link

murk-sy, 2025-10-05:
@3d-gussner Another instance of this type of issue, makes the same noise as well:
https://www.reddit.com/r/prusa3d/comments/1ny2neh/why_is_my_xl_do_this_bs/