prusa3d/Prusa-Firmware-Buddy #4680: [QUESTION] Tool map screen

ansonl, 2025-06-09:
### Question

Hi, what gcode is required to have a tool show up on the PrusaXL tool mapping screen?

I have a post processed gcode file that I have added additional toolchanges to print certain features in a different tool and the PrusaXL tool mapping screen only shows the tools that the model with sliced with in PrusaSlicer.

[image]

When the layer with the other tool index is run, an error shows up.

[image]

So far I have found the following generated gcode that references the used tools T0, T1, T2, T4 (if these are the used ones in the slicer) in PrusaSlicer but adding them into a gcode file that sliced with 1 tool still only shows a single tool for tool map.

```
; inform about nozzle diameter
M862.1 T0 P0.4 A0 F0
M862.1 T1 P0.4 A0 F0
M862.1 T2 P0.4 A0 F0

M862.1 T4 P0.4 A0 F0

; turn off unused heaters

M104 T3 S0

T1 S1 L0 D0
T2 S1 L0 D0

T4 S1 L0 D0
; select tool that will be used to home & MBL
T0 S1 L0 D0

; purge tool (for each used tool)
; not copied here because it is long

; turn off extruder heaters
M104 T0 S0
M104 T1 S0
M104 T2 S0

M104 T4 S0
```

CZDanol, 2025-06-10:
Hello, it is determined by some metadata, like material types or materials used I believe. Generate yourself some non-binary XL multitool gcode and look at the `;`s at the end of the file

ansonl, 2025-08-11:
I've looked at the gcode metadata and tried including the metadata from a multicolor print but the PrusaXL still shows only the sliced tools for tool mapping.

Can any of the Prusa team please provide some detail on what metadata fields are needed to show up on the pm firmware tool mapping screen?