prusa3d/Prusa-Firmware-Buddy #4697: [ENHANCEMENT] Allow X Y Z step calibration, where E steps are in menu.

kenour, 2025-06-21:
### Printer model

MK3.5S

### Describe the enhancement

Related to [BFW-7094], allow the X Y Z steps to be calibrated the same way E can. Even if you only allow +/- 1%, I've found this the best way to get models bang on the correct size.
I don't want to have to upsize the model (which I have to while it's still broken in firmware [BFW-7094]), as this causes issues with features not appearing where expected, especially in the Z axis.
The reason I want a hardware solve is so that an object 100mm tall, is 500 x 0.2mm layers exactly. That way I can plan out features perfectly in CAD, in multiples of 0.2 as I always do, then being able to slice and print without having to tweak anything in software to get an on size part.
I understand this will result in a reduced print volume (albeit small), but this should be fed back after changing to the bed size can be changed in slicer.

### Expected functionality

I guess the expected functionality is to have a model come out the expected size. This isn't a material shrinkage issue, this is a hardware issue. Fixing in software isn't the correct solution.
My MK3.5S doesn't just print undersize, the X and Y axis are different factors, so a linear scaling factor doesn't solve the issue.
I was noticing weird issues with larger threads for models I was printing in CF (M30mm+), and when measuring them noticed the threads were oval.

mike-bee, 2025-06-23:
I also have created a related ticket for the XL (#4681). Once the underlying issue has been solved, setting the steps via menu would be really nice to have, though a bit more complex with multiple toolheads.

murk-sy, 2025-06-23:
I would generally recommend against changing number of steps for X and Y. Steppers have a known number of steps per rotation and pulleys have a known diameter, and it's not like they change. Changing steps means you're also changing the spacing between extrusion lines, not just the overall size.

Having the option available could cause users (that definitely shouldn't do that) adjusting steps based on measured printed parts, which used to be the predominant advice some years ago, and I'm sure there's still some that dig it up nowadays.

As you mentioned, hardware problems shouldn't be solved in software. I'd recommend closely inspecting (and ideally comparing) the belts for wear and tension. If you're printing ovals instead if circles you need a service, not firmware corrections.

kenour, 2025-06-23:
I suggested 2 safeguards to ensure inexperienced users don't make boo boos. Hiding it in the advanced menu (where you can change the extruder steps), and limiting the amount it can be updated (+/- 1%). This isn't software, it's firmware, which I see as more of a hardware calibration.

My machines are well maintained, but even so, this is calibrating out the last 1% error to ensure models are in spec. This is critical to me as I print tool guides, dragons and crap.

I have been doing this for years, since I first noticed my prints were slightly undersized. I haven't needed to change the scaling factor since the initial correction, which to me indicates each printer probably has their own slight variation that should be able to be corrected in hardware (via firmware), not scaling models in software (which is so annoying when I forget to do it now, for EVERY bloody model, instead of once).

I think I'll upload my model and method to printables, see what results people get. See how similar/different people's results are.

Regarding belts and steps etc, this doesn't matter. I'm simply adding a HW scaling factor, so the printer actually prints what it's sent. My prints have been perfectly on spec for maybe 6 years since I initially incorporated this fix. I do routinely service and calibrate my machines, to ensure the parts are in spec, otherwise they wouldn't be fit for purpose 🤷🏼‍♂️

Anyway, I'm going to continue using my method (which I've proven works, consistently and flawlessly for years), once the firmware is unborked that is... 🙄 I really look forward to re-slicing all my production models again 😳

kenour, 2025-10-01:
> Thank you for your contribution to our project. This issue has not received any updates for 60 days and may be considered "stale." If this issue is still important to you, please add an update within the next 7 days to keep it open. Administrators can manually reopen the issue if necessary.

[image]