prusa3d/Prusa-Firmware-Buddy #4703: [BUG] Core One + MMU - "Filament already loaded"

nschllr, 2025-06-25:
### Printer model

Core One

### Firmware version

6.3.3

### Upgrades and modifications

MMU (lite)

### Printing from...

USB-Drive, Connect, Prusa-Slicer

### Describe the bug

Every time I start a print, the Core One states that there is already filament loaded and asks whether I want to keep the current filament or unload it — even though there is no filament loaded (this issue started after upgrading to the MMU). However, both filament sensors (Nextruder and SuperFINDA) show that no filament is detected.

I already tried the fix suggested in the Knowledgebase (unloading via the LCD), but the problem persists. It also happens after restarting the printer, and I have calibrated both sensors multiple times.

### How to reproduce

Start a print with enabled MMU on Core One and the printer states that there is already filament loaded

### Expected behavior

Start a print with enabled MMU on Core One and the printer checks if the SuperFinda or Nextruder sensors detects filament, if not start the print

3d-gussner, 2025-06-26:
@nschllr Thanks for the update. I have been trying to reproduce your issue for some time and didn't get the result you had.

3d-gussner, 2025-06-26:
Please consider to close the issue

nschllr, 2025-06-26:
Have you also tried reproducing the issue with the side filament sensor disabled? If not, that might be the cause — in which case it would be a bug. If you have already tested it that way, then it’s probably just a weird state on my printer, and I’ll go ahead and close the issue.

3d-gussner, 2025-06-27:
@nschllr Add your issue to an internal issue tracker. Thanks for reporting.

VH-fixer, 2025-07-25:
I am using the MMU3 on the Core One and have turned off the side filament sensor.    Firmware 6.3.4.  After printing one job, the next job reports the filament is loaded when nextruder and Pinda show not loaded.   I select YES to unload in Prusa Connect message.  It soon senses it is unloaded and continues.      Then it goes on and prints with the MMU3.    To reproduce, You may have to print 1 job first and then it shows up on the succeeding prints.

micag001, 2025-09-16:
Exact same issue Core One MMU3 on firmware 6.3.4. Issue only happens when I have the side filament sensor deactivated (because of MMU3). Enabling the side filament sensor eliminates the issue. Thank you for the tip.

lordjaxom, 2025-10-23:
Same here, on firmware 6.3.4. Side filament sensor disabled but physically attached. Printer reports "filament already loaded" although the sensor shows OFF on the LCD.

3d-gussner, 2025-10-24:
@lordjaxom The side sensor needs to be enabled in the 6.3.4 version with the MMU. The developers are working on a solution.

micag001, 2025-10-28:
Hi, as has been said, the issue does not occur with the side sensor enabled in the firmware, and everything works as intended.

closed 2026-10-12 by github-actions[bot]