prusa3d/Prusa-Firmware-Buddy #4708: [BUG] Quitting a print on MMU3 forces user selection of filament to unload

Samuel1698, 2025-07-06:
### Printer model

MK4

### Firmware version

6.2.0-alpha2+8702

### Upgrades and modifications

MMU3

### Printing from...

PrusaConnect

### Describe the bug

When printing remotely, if I hit the Quit button on PrusaConnect because I notice an issue, the printer forces me to select which filament is loaded (despite the fact that the filament didn't change from the time it was loaded - single color print on mmu3) with no way to select that on PrusaConnect. It will idle at that temperature forever unless I physically go and press the button once. 

Of note is that PrusaConnect correctly identifies which filament is loaded, but the printer does not? 

### How to reproduce

Start print
Go to Connect
Hit "Quit"
Printer status goes to "Busy" and LCD screen will show a selection

Samuel1698, 2025-07-18:
If I "quit" the print on the LCD screen mid-print it also prompts me to select the filament to unload. Never noticed it was a problem then. But being unable to do so remotely is still an oversight

3d-gussner, 2025-07-18:
@Samuel1698 I can reproduce your issue and created an internal ticket.

closed 2026-02-02 by github-actions[bot]