prusa3d/Prusa-Firmware-Buddy #4710: [BUG] M84 at end g-code not disabling motors

steelydev, 2025-07-07:
### Printer model

core one

### Firmware version

6.3.3+10442

### Printing from...

Prusa Link

### Describe the bug

At end of print some motors, for sure the z-motors are active.  Using the menu Control->Disable motors works.  End g-code does not disable.  Changed Prusa default value of M84 X Y E to M84 X Y Z E and that did not disable the motors.  Also tried and failed with simply M84.

### How to reproduce

Print an object.

### Expected behavior

Motors should disable at end of print when the M84 code is sent.

audkar, 2026-10-12:
The update is, that I use Prusa the least from my printers, because after printing I have to manually disable motors to avoid motors noise.

What is this new strategy, that clients have to report every few months that items manufacturer still haven't fixed their issues? Or otherwise tickets get closed? Shouldn't this be other way around, and manufacturer give an updates?

steelydev, 2026-10-13:
It is outrageous that this has not been addressed and the system keeps trying to stale it!!!

Prusa please respond.