prusa3d/Prusa-Firmware-Buddy #4714: [BUG] LCD gets dark screen randomly and remain dark, but the printer is still working fine.

ernstc, 2025-07-12:
### Printer model

Core One

### Firmware version

6.3.3

### Upgrades and modifications

None

### Printing from...

USB drive, Prusa Connect

### Describe the bug

I have assembled my Core One Kit and after few days, I am having this issue where the screen gets dark and remains dark until I reset or restart the printer from the main power switch. This happens randomly, sometimes event at startup after showing the progress of the boot loader. But when the screen is dark it seems that the printer is still working, so it seems to be just a problem of the screen.

### How to reproduce

It happens randomly.

### Expected behavior

The screen don't gets darker. If it is a feature of a kind of screen saver, there is no way for getting back to the normal screen with printer info and commands.

CZDanol, 2025-07-12:
Hello, this does not look a firmware issue. Either your display or cable is faulty, or for example the connector might be not pushed in fully. Please contact our support.

closed 2025-07-13 by ernstc

3d-gussner, 2025-07-15:
@ernstc I had a similar issue flickering LCD and going dark when powering up on my MK4 and by setting the refresh rate to LOW this is gone.
The new firmware should solve some users have with their LCD screens like you.