prusa3d/Prusa-Firmware-Buddy #4715: [BUG] Cannot use any toolhead filament sensors if any one of them is broken, due to calibration lock-out logic

meermanr, 2025-07-13:
### Printer model

XLIS 5T

### Firmware version

6.2.5+8912

### Upgrades and modifications

XL enclosure upgrade

### Printing from...

N/A (USB drive via Prusa Connect)

### Describe the bug

One of the toolheads on my 5-tool XL has a faulty filament sensor. At some point in my troubleshooting I tried to recalibrate it using "Filament Sensor Calibration" test, which confirmed my suspicion that it is indeed faulty (no detection). This made it impossible to complete the calibration, so I reset the printer via the RESET button. Now I cannot use **any** of the filament sensors (both side sensor and toolhead sensor). At the start of each print I see "Warning: Please complete Calibration & Tests before using the printer." (Which I cannot do while it has a fault).

[image]

I cannot manually reenable any of the sensors, presumably because entering the "Filament Sensor Calibration" test cleared their data:

[image]

[image]

I feel this is a bug, because I cannot make use of the spool join or filament runout features on the remaining toolheads. Had I not initiated the calibration I would have been able to simple disable the faulty filament sensor and worked around this until it was convenient to address the fault. After all, a big advantage of having mulitple heads is that you have redundancy so you can keep producting parts even if a fault develops in one.

### How to reproduce

1. Have a faulty filament sensor (e.g. disconnect one)
2. Enter "Filament Sensor Calibration" test
3. Fail to complete the test due to the faulty sensor
4. Accept defeat, make a note to address this fault when production is finished, and RESET the XL to get back to producing parts
5. Find that **all** filament sensors are now disabled, even those that were previously working

### Expected behavior

Aborting the "Filament Sensor Calibration" test via RESET should leave the system in the state it was before I started the test: filament sensing is enabled and the working sensors act as exected.

Alternatively, the calibration test allows me to skip some sensors that seem to be faulty, and disables them in the Filament Sensors menu. That way the system is in a better state (more consistent and true to reality) after calibration than it was before (where a faulty sensor was still enabled)

### Files

Firmware:

[image]

Filament sensors menu showing that filament sensing is globally disabled:

[image]

Calibration menu showing that "Filament Sensor Calibration" has not been completed (i.e. launch it before the RESET wiped my settings):

[image]

meermanr, 2025-09-15:
Responding to keep this issue open, as I suspect it’s fairly easy to address: delay wiping calibration settings until the end where new settings are being saved

3d-gussner, 2025-09-15:
@meermanr Sorry for the late answer. Can you tell which FSensor fails?

A workaround:
- Disconnect the faulty tool
  - Please ensure that you have 1 to 4 in correct order and the broken one (until you fix the sensor) on the 5th position (disconnected)
- Reset your printer and it should detect 4 tools only and be able to run the calibration and tests.
  - When you have to switch the faulty nextruder to 5th position, it is recommend to run the whole calibration again. Especially the docking and tool offset calibration.

Hope that helps for the time being.

3d-gussner, 2025-09-16:
@meermanr Glad to hear that you could fix your printer. Created an internal ticket.

closed 2026-09-15 by github-actions[bot]