prusa3d/Prusa-Firmware-Buddy #4720: [FEATURE REQUEST] Core One chamber as filament dryer

dziedzic99, 2025-07-16:
### Printer model

Core One

### Describe the feature

Core One chamber can heat up significantly using the heat bed. It can also replace air inside the chamber using its exhaust fans. Therefore it's a filament dryer, with the extra feature of 3D printing.
Please test this and add a feature to use the chamber as a dryer.

### Expected functionality

I would love a menu option to enable filament drying with different temperatures for different filaments. The feature should prompt user to configure the vents and put the heatbed in a proper position. Then the chamber would heat up and periodically vent the air.
This would require testing on whether the spool(s) should be placed under or on the heatbed (maybe with a spacer in between ?) and an open loop solution as there is no humidity sensor available.

m-cas, 2025-07-17:
What if you would offer it after a print is completed? Then it would be just a little more energy to conserve temp but the main "warming up" is done by the printing... eco balance would be probably better than an external dryer... Ideally it would be a selectable option (offer filament drying after printing Y/N). 
Possibly adding a "standalone dryer" mode for the less eco friendly (or those that do not have an external dryer) with the proper warning about energy waste, etc etc

dziedzic99, 2025-07-17:
@3d-gussner 
Thank you for your reply. I acknowledge that the power efficiency of this solution is expected to be poor. I kept the chamber heated up to 50C and it was consuming about 90W which is subpar compared to any single spool filament dryers. But if you consider you can stash 5 filament spools inside  it doesn't look bad.
Please keep me posted should you have any news.

closed 2025-09-24 by github-actions[bot]