prusa3d/Prusa-Firmware-Buddy #4721: [BUG] Worse ironing in the MK4S vs MK4, and poor adjustment of the ironing height relative to the bed

I3estOFF, 2025-07-17:
### Printer model

MK4S Kit

### Firmware version

6.2.4+8909

### Upgrades and modifications

MMU3

### Printing from...

USB drive

### Describe the bug

[image]

After upgrading to the MK4S and printing the same parts, there's a noticeable difference in ironing quality. On the MK4S, it looks different across various parts of the bed. In most areas, the quality is very poor. It seems like the ironing in the MK4S is not adjusted to the height of the bed because in some spots, it barely touches the print, while in others, it's actually scraping filament off the print. I tried changing the ironing option, for example, from 15% to 10% or 5%, but the issue isn't that the entire bed is bad — it's that the ironing quality differs in each part of the bed.

Here’s a comparison of the ironing quality between MK4 and MK4S.
[image]

### Expected behavior

It should iron evenly across the entire surface of the table, or without such a big difference

I3estOFF, 2025-07-17:
Okay i'll test it!

3d-gussner, 2025-07-29:
@I3estOFF Did you have time to test the older firmware version?

I3estOFF, 2025-07-29:
Sorry, I haven't had time to test lately. I just ran a test print with ironing, and another issue has come up: some objects don't get filament after the previous one has been ironed, which results in gaps. But this might not actually be a firmware problem, it could be an issue with my extruder?

[image]

[image]

I3estOFF, 2025-08-19:
I tried to test it, but new issues kept coming up. Specifically, after ironing the filament would get stuck in the extruder, so when the MMU3 tried to retract it, the filament snapped inside the extruder. Combined with ironing each part one by one (first printing the part, then ironing, then printing the next one), this sometimes caused certain parts to be missing their top layer, with the ironing happening in mid-air. I’m using Prusament (so it’s not just some random filament), and after ironing the last layer the filament snapped in the extruder, which then prevented the next object from being printed because the extruder was clogged and couldn’t push the broken piece through.

To me, this looks like the nozzle temperature is too low after finishing the ironing for retracting the filament back into the MMU.

[image]