prusa3d/Prusa-Firmware-Buddy #4723: [FEATURE REQUEST] Allow user to select a simple X/Y homing, avoiding the serial head bangs issue

zapta, 2025-07-20:
### Printer model

CORE One

### Describe the feature

**TL;DR**, **Allow users to select** a simple homing mode that performs a **single X head bang and a single Y head bang**, avoiding the repetitive banging issue reported by many users.

This issue is a carry over from the **Printing head calibration going wild** thread in the Prusa forums 

https://forum.prusa3d.com/forum/prusa-core-one-assembly-and-first-prints-troubleshooting/printing-head-calibration-going-wild/paged/2/#post-758279

Many CORE One users currently face the issue of repetitive print head banging during X/Y homing, having ~10 bangs on a good days and in other cases 100+ successive bangs (I experienced it personally). 

Thisfeature request is to allows users to turn off the precise X/Y homing for print resume and use instead a single X and single Y bang homing that is good enough to start a print. 

The print-resume-after-power-failure is a feature that most users don't need and don't use and thus do not need that successive head banging policy.

3d-gussner, 2025-07-21:
@zapta This is a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4723 or at least similar to it. I am closing this one. Also added this ticket as a sub-ticket to the existing one.

closed 2025-07-21 by 3d-gussner

zapta, 2025-07-21:
@3d-gussner meant duplicate of #4582