prusa3d/Prusa-Firmware-Buddy #4726: [BUG] Phase stepping calibration produces inconsistent results

nbun, 2025-07-23:
### Printer model

Core One

### Firmware version

6.3.4

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

The results of the phase stepping calibration are unclear and inconsistent. When I run the calibration multiple times, I get results anywhere from 20% to 80% reduced vibrations. This implies a massive difference, which I'm not sure is actually the case.

Part of the issue might be that it is unclear what is used as the baseline. Does the calibration use previous calibration data and try to improve that (multiple calibration runs should inevitable lead to a smaller reduction in vibration because the calibration is nearing optimal results) or does the calibration always use default values (this would mean that the calibration is heavily influenced by external factors leading to inconsistent results). Put differently, should it make a difference in results when I reset the phase stepping calibration between two calibration runs?

A fix depends on which of the above is true. I would prefer an absolute value (similar to input shaping where I always get a absolute Hz value) as result. If the calibration always compares to default values, it needs to be made more robust against unpredictable factors that skew the calibration.

### How to reproduce

Run the phase stepping calibration

### Expected behavior

Repeated phase stepping calibration produces similar results (within ~5%)

nbun, 2025-07-25:
This video also shows the problem: https://www.youtube.com/watch?v=R0NEuJtWj-k

bkerler, 2025-08-19:
If you run the input shaping directly prior to the phase stepping calibration, does it show the same behaviour ?

bkerler, 2025-08-20:
Guess someone from Prusa HW needs to answer these questions. All I can say that running the input shaping calibration has a major effect on phase stepping calibration due to the homing process as it seems. Nevertheless, for me input shaping is pretty consistent but phase stepping hasn't been but that might be related that the phase stepping only shows optimization improvements and not the actual values. If you run the phase stepping via gcode, you should get much more consistent values via serial terminal.

nbun, 2025-08-20:
@danopernis You mentioned in #4461 to open a new issue regarding phase stepping calibration problems. Could you clarify how the calibration is expected to work and how to interpret the results?

Prusa-Support, 2025-09-13:
These are legitimate questions that require clarification from the developers and product managers.

The Help article, though, hints at belt tension as a possible cause of failure, which seem to suggests that the belt tension can affect the measured value, at least to some extent.
https://help.prusa3d.com/article/phase-stepping-core-one_914247

For a more precise reading, you may want to try further turning the [belt tension (consider trying the new firmware's method)](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-alpha#:~:text=accessible%20via%20Settings%20%E2%86%92-,Manual,-Belt%20Tuning%2C%20begins) and [cleaning and lubricating rods and linear guide](https://help.prusa3d.com/article/regular-printer-maintenance-core-one_829710) if necessary.
Then you may try to check the calibration consistency after a [calibration selective reset (new feature)](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-alpha#:~:text=These%20categories%20include%3A-,Calibrations,-%3A%20All%20the%20calibrations) to start every time from stock values.

Michele Moramarco
Prusa Research

nbun, 2025-09-15:
> The Help article, though, hints at belt tension as a possible cause of failure, which seem to suggests that the belt tension can affect the measured value, at least to some extent. https://help.prusa3d.com/article/phase-stepping-core-one_914247

A quick note: The QR code on the phase stepping calibration page points to the generic accelerometer article, not the article specific to phase stepping. 

> For a more precise reading, you may want to try further turning the [belt tension (consider trying the new firmware's [...]

I already tuned the belts using the new method. As requested, I lubricated the rods and reset the printer's calibration data. This procedure takes roughly 30 minutes (including the first-time setup) and I don't have the time required to reset and re-calibrate the whole printer a few times. Nevertheless, repeating only the calibration still produces inconsistent results.

3d-gussner, 2025-11-17:
> [@Prusa-Support](https://github.com/Prusa-Support) From the commit history it seems that the Core One L firmware uses a new algorithm for the phase stepping calibration. Will these changes make it into a Core One release or do we have to wait until the firmware is unified again?

The commit should also be in the new firmware release for the CoreOne and other printers (which support phase stepping).