prusa3d/Prusa-Firmware-Buddy #4732: [BUG] Input Shaper causing layer shifts

imax9000, 2025-07-25:
### Printer model

CORE One

### Firmware version

6.3.3, 6.3.4

### Printing from...

USB cable

### Describe the bug

I think I'm hitting a very unfortunate interaction between Input Shaper and printing circular objects. It looks like sometimes the changes Input Shaper injects can cause belt skips/motor misbehaving.

Reproduces on 6.3.4 too, with freshly tuned belts.

### How to reproduce

Set Input Shaper parameters as follows (this is what calibration has generated on my printer): 
* X axis: ZVD, 45Hz
* Y axis: MZV, 51Hz

With that done, run the included [`repeat.gcode.txt`](https://github.com/user-attachments/files/21436091/repeat.gcode.txt). It will perform 50 times a problematic move I extracted from a sliced file. Despite feed rate not changing during the move, the printhead moves in bursts, and sometimes doesn't move to the correct position (usually with a squeaking sound coming from one of the motors).

At the end it will move the printhead into each of 4 corners, if no layer shifts occurred - you may hear maybe a gentle bump in some corners. OTOH, if there were layer shifts - you will hear squeaking as it happens, and at the end it will crash into some of the edges.

### Expected behavior

If I set Input Shaper to Null for both X and Y,  the move happens much more smoothly and without layer shifts.

### Files

[repeat.gcode.txt](https://github.com/user-attachments/files/21436091/repeat.gcode.txt)

Problematic move extracted from a sliced file: [move.gcode.txt](https://github.com/user-attachments/files/21436096/move.gcode.txt)

A script that generates a torture test using `move.gcode`:  [gen.sh.txt](https://github.com/user-attachments/files/21436107/gen.sh.txt)

imax9000, 2025-07-31:
Hm, while input shaper seems to substantially aggravate the issue, it is probably not the root cause.

I've got another failed print today, so decided to check the printer again. Belt tension has dropped a little bit, but still within the ranges recommended by the new version of https://belt.connect.prusa3d.com/. I've disabled input shaper and reset phase stepping to defaults, but my torture test was still failing, until I've dropped acceleration from 7000 all the way to 4000.

So it's most likely either a mechanical issue with my printer that I need to sort out, or the default settings in the slicer are overly optimistic to begin with.

[torture_test.zip](https://github.com/user-attachments/files/21534479/torture_test.zip)

imax9000, 2025-08-15:
I'm not sure it's a purely mechanical issue now. It seems that the printer is struggling with curved toolpaths that consist of many tiny segments:

* Circle using four 90° arcs - no problem
* Circle using 360 linear segments - very bad

"Avoid crossing perimeters" option (#4573 ) is particularly prone to generating such toolpaths, especially with objects that have circular/curved surfaces. "Arc fitting" in theory should allow to make it better, not worse (#4737), but I did not inspect the generated g-code to compare the number of commands emitted by the slicer.

I've noticed that LCD UI gets very laggy too. I don't know if that's a cause or just another symptom: CPU not keeping up with driving the motors might drive them incorrectly (I can be very wrong on this one, as of now I've no idea how the path from g-code to motor signals works), or the problematic code paths just happen to also be more computationally expensive.

In the same vein, Input Shaper might be making the problem more frequent either due to how specifically it changes motor signals, or by simply putting additional load on the CPU.

Prusa-Support, 2025-09-13:
There is a new final release available for Prusa Slicer ([2.9.3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.3)), and a new firmware pre-release ([6.4.0-alpha](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-alpha)).
Please check out and possibly test our stock profiles - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4573#issuecomment-3288285067.

Michele Moramarco
Prusa Research

imax9000, 2025-09-13:
> new firmware pre-release ([6.4.0-alpha](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-alpha)).

I'm still filming, but it seems to have made the problem much worse. Is it built with the same optimization and debug flags as a regular release?

imax9000, 2025-09-13:
Ok, so what I did:

1) Flashed 6.4.0-alpha firmware
2) Tuned the belts with the new wizard
3) Reset Input Shaper settings to factory defaults
4) Ran `torture_test_x500_acceleration-2500.gcode` from `torture_test.zip` I've uploaded in [this comment](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4732#issuecomment-3140652135)
5) Ran `torture_test_x500_acceleration-7000.gcode`
6) Rolled back to 6.3.4
7) Without changing any settings at all, ran `torture_test_x500_acceleration-2500.gcode`
8) Ran `torture_test_x500_acceleration-7000.gcode`

I've stopped both runs with max acceleration 7000 early, to not have the poor thing needlessly suffer.

And I repeat: the move that it's repeating 500 times was copy-pasted from an actual print gcode generated by PrusaSlicer.

Here are the videos: https://www.youtube.com/playlist?list=PLAomIv-HUJevdW4XWB_aWBiJD_lTsWjbA

My best-fitting hypothesis is still that the task responsible for driving the steppers gets CPU-starved and sends mistimed signals. In that case lowering acceleration limit helps by reducing the max velocity that the printhead can reach over the travel distance, meaning that gcode commands are being processed at a lower rate, leaving more CPU time available to the task that drives the steppers.

imax9000, 2025-09-15:
> [@imax9000](https://github.com/imax9000) can you do me a favor and test the gcode but set the Z height to the same value, so that the z-height is all the same ? Would be interesting if you then see the same issues ...

If I don't forget 😅 Right now I have a print in progress until tomorrow late afternoon.

I already tried generating a circular movement (w/o Z height changes) from tiny linear segments and it also triggered the same issue at certain segment sizes, but it wasn't a straightforward "more segments - more worse". Might be also good to try that again, as well as splitting a linear path into segments.

In the meantime, feel free to grab my scripts from https://www.printables.com/model/1385614-core-one-layer-shifts/files and try different options yourself.

3d-gussner, 2025-10-06:
@imax9000 Thanks for the time and effort to look into that. The developers are also working on similar/related issues but at the moment I can't add an internal ticket to this github issue.