prusa3d/Prusa-Firmware-Buddy #4735: [BUG] possible bug with  changing filament during print and generall the change filament procedure

AvariCe-git, 2025-07-28:
### Printer model

Core One

### Firmware version

6.3.4

### Upgrades and modifications

Idk if the obxidian nozzle is considered an upgrade

### Printing from...

Usb drive

### Describe the bug

The bug happens where ever the 'change filament' procedure occurs. 
It unloads the pre existing filament well, and then it shows the dialogue where it asks if the unload was successful. 
If I choose yes and remove the filament, then it shows '10% loading' and after a few seconds, it considers the loading as a success and coontinues working. That happens irrespective of whether I I've loaded a new filament. 

If I go the filament menu and do the unload procedure and the load procedure on their own, it works as expected. 

### How to reproduce

1. Print something with filament change in the gcode or choose the option filament  --> change filanent

### Expected behavior

After unloading the first filament, it should show 0% on the loading screen and wait for me to put another filament. 
Also, I've made of bunch of filament sensor calibrations, and the problem is still persists, and the filament sensor works exactly as it should

Prusa-Support, 2025-09-25:
As far as I can see, on the last firmware, the problem is not reproducible, except when
**[a] the filament sensor is actually disabled**,
**[b] not calibrated (N.A. status)**.
In the latter case (b) the printer will basically behave in the same way as the filament sensor was not enabled (a).

Now the filament change and a few processes are more "descriptive" too (_parcking, ramming, inserting, loading, purging, unparcking_), and the progress bar (and %) is now more "indicative" of the ongoing process progress.

Please try FW 6.4.0-RC, the final release is expected soon.
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-RC

Michele Moramarco
Prusa Research

3d-gussner, 2025-11-19:
Please try firmware [6.4.0-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-RC2) and report back.