prusa3d/Prusa-Firmware-Buddy #4737: [BUG] Displaced layers with Arc Fitting enabled for rounded objects

ErichStyger, 2025-07-30:
### Printer model

Prusa Core One

### Firmware version

6.3.3, 6.3.4, 6.3.5-alpha

### Upgrades and modifications

factory build

### Printing from...

USB drive, Prusa Link, Prusa Connect

### Describe the bug

Printing models with rounded corners or round models, the results are nozzle crashes and displaced x/y layers. 
Disabling 'arc fitting' has been identified as workaround.

### How to reproduce

Print a model with round corners or with circular structures, e.g. attached files.

Observations:
- noticed sometimes smaller (1 mm) layer displacements
- hearing some strange noise during printing
- print head is crashing into already printed structures, causing x/y position displacement of the head
- problem mostly occurs in the lower layers, but not always. And not always in the same place

Image below shows the issue observed in one corner of the model:
[image]
The printer seems to take a 'shortcut' with arc fitting enabled: instead printing the outside round corner, it prints multiple straight lines inside on top of each other: this then causes the nozzle to crash into it, causing x/y displacement.

Workaround:
As the problem seems to be related to rounded structures, I tried disabling the 'arc fitting' which then printed the parts properly.
[image]

### Expected behavior

The printer should print round objects or rounded structures with Arc Fitting enabled.

### Files

Model is from https://www.printables.com/model/944916-modern-rectangular-key-tray
Attached the files with arc fitting disabled (working) and with arc fitting enabled (not working).

[arc fitting problem.zip](https://github.com/user-attachments/files/21501600/arc.fitting.problem.zip)

3d-gussner, 2025-07-30:
@ErichStyger Please close this issue here and add your comment in #4573.

ErichStyger, 2025-07-30:
I don't believe it is the same issue. In my case neither ACP or STRUCTURAL fixed the problem, only *disabling* Arc Fitting helped. There is as well a similar report around the input shaper here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4732 
Unless Prusa engineering confirms that all these are the same issue and have the same root cause, we should not have these marked as duplicates.