prusa3d/Prusa-Firmware-Buddy #4742: [ENHANCEMENT] Add home calibration to pause / tune menu while printing (as there is no collision detection anymore)

Mr3Deee, 2025-08-03:
### Printer model

XL Single / Multi tool

### Describe the enhancement

As there is no collision tracked anymore since a while (and I am not sure if someone is working on this), a potential collision within a long print, ruins the print, even if you see it while it happens.

As such my suggestion would be to extend the printing "tune" menu in pause mode with a "calibrate position". That way the print might still be savable.

==> Please add an option to recalibrate home from the printing "tune" menu.

### Expected functionality

Recalibrate home from the pringing "Tune" menu.

murk-sy, 2025-08-04:
Duplicate of:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4277

CZDanol, 2025-08-06:
Hello, homing calibration can result in homing origin shift, so it is not possible to calibrate homing while printing.

If you mean not calibrate homing, but just home during print, then yes, that is already covered in #4277, and thus I'm closing this as a duplicate.

closed 2025-08-06 by CZDanol