prusa3d/Prusa-Firmware-Buddy #4747: [BUG] Prusa XL, Load all filaments for the print, crashes the printer

Mr3Deee, 2025-08-08:
### Printer model

XL Multitool

### Firmware version

6.2.6-alpha+8916

### Upgrades and modifications

None

### Printing from...

PrusaConnect

### Describe the bug

- Start printing a model with multiple tools.
- Ask the printer to load all filaments
- Begin the process and insert the filament.
- When the screen pops up that a filament insertion has been detected, press QUICKLY the "continue" button via touch.
==> Crashes with a reboot screen that a printerhead crash has occurred.

### How to reproduce

- Start printing a model with multiple tools.
- Ask the printer to load all filaments
- Begin the process and insert the filament.
- When the screen pops up that a filament insertion has been detected, press QUICKLY the "continue" button via touch.
==> Crashes with a reboot screen that a printerhead crash has occurred.

### Expected behavior

It simply inserts the filament. There was no tool head crash, all which has happend was a doubleclick on the display button via touch.

### Files

Seems simple enough to see the crash. It has happened to me already multiple times.

3d-gussner, 2025-08-08:
@Mr3Deee Thanks for reporting.
>When the screen pops up that a filament insertion has been detected, press QUICKLY the "continue" button via touch.

- Does it work when you "wait" and touch to continue?
- Does it work when you "wait" and click on knob to continue?
- Does it work when you QUICKLY click the knob to continue?
  - Does it work when you QUICKLY double click the knob to continue? 
- Was the double click "touch" that causes the printer to crash?

Mr3Deee, 2025-08-08:
I did "trigger" the crash 3 times now by accident via touch. I didn't try the other methods as a crash is usually pretty heavy and I do not really want to see the progress bar and so on. Once I tried to turn off and on the printer thereafter and got a message about another startup program failure (I unfortunately didn't take a screenshot back then though).

Having said that - I suppose that you can give it a try quickly? If not I can see that I check it once my current print is finished.