prusa3d/Prusa-Firmware-Buddy #4754: [BUG] Absorbing heat % wrong value

dmognaschi, 2025-08-11:
### Printer model

Core ONE

### Firmware version

6.3.4

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

Printing on flex material.
Ambient temperature: 34° C (summer).
Starting print, the 'absorbing heat' phase never ends and the percentage is huge (maybe a wrong cast of a negative number). See picture.
[image]

paynterf, 2025-09-03:
Any progress on this?  Had the same thing happen to me printing PP with heat bed set to zero.  'Absorbing Heat' shows the same (impossibly large) percentage as in OP's photo, and never ends.  Have to select 'Skip' to proceed

bkerler, 2025-09-04:
@paynterf can you please state exactly what you did and when this happens? I cannot reproduce it on my setup and in order to do the PR properly, it would be really helpful.

paynterf, 2025-09-04:
I just used the Prusa default settings for PolyPropylene, which has the bed temp set at zero.  When I attempt to print this, I got the 'large' heat soak percentage.  I'm in the middle of a long print right now, but I'll try to duplicate the problem in a way that you can reproduce