prusa3d/Prusa-Firmware-Buddy #4761: [FEATURE REQUEST] Core one doesn’t need to home when unloading filament at end of print

mhornsby, 2025-08-15:
### Printer model

Core one

### Describe the feature

With no prints in progress the Core One tries to home when unloading filament. Homing is not necessary. It can lead to problems if the homing fails because the filament will not unload and you end up filament stuck in the printer

Cbokar, 2025-08-17:
I would strongly appreciate that. But also for loading filament. One tap x and y are enough.
( one tap is possible. When the head is in parking position it taps once to the side.)

The permanent banging is really very annoying.

G28 i.   Helps a bit before mbl.

mhornsby, 2025-08-26:
Yes its a dup