prusa3d/Prusa-Firmware-Buddy #4762: [BUG] Core One FW6.3.4 LED Dimming is not working

RPDengel, 2025-08-17:
### Printer model

CORE ONE

### Firmware version

6.3.4

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

Hi!
Since the update to 6.3.4 the dimming of the LED is not working. I've chosen "chamber dimming: always" and the LED is not dimming. Not at Printing, also not in idle.
cheers,
Ralf

3d-gussner, 2025-08-20:
Internal test confirmed on some printers the issue on others it wasn't.

One solution was to partly to factory reset the printer. If anyone of you could confirm that this solves the issue would be great.

Zemistr, 2025-08-20:
> Internal test confirmed on some printers the issue on others it wasn't.
> 
> One solution was to partly to factory reset the printer. If anyone of you could confirm that this solves the issue would be great.

Will this clear the print time counter?

3d-gussner, 2025-08-20:
@Zemistr You can choose what to keep and what to reset. So if you keep the Statistics the print time counter should not be reset.

RPDengel, 2025-08-20:
@3d-gussner I've have done the factory reset. 
By now the dimming was working as expected, with 6.3.5-alpha+10514
Ive tested the "always" option.

Zemistr, 2025-08-20:
> [@Zemistr](https://github.com/Zemistr) You can choose what to keep and what to reset. So if you keep the Statistics the print time counter should not be reset.

There was no option to choose whether or not to keep the print time counter.
So I lost all print time, which was very important to me for a review I'm filming right now.
Dammit, I was stupid to even do the factory reset...
Thanks a lot 🫡

3d-gussner, 2025-08-21:
@Zemistr 
>There was no option to choose whether or not to keep the print time counter.
So I lost all print time, which was very important to me for a review I'm filming right now.

I have the pre-release of the firmware installed on my printer and didn't check with current public version, when I wrote the comment and instructions.

Normally I switch between the public and development firmware's, but this time I forgot about it. Really sorry for that 🙏  

Side note: I hope my proposal to be able to backup/restore or read and write some saved values will be implemented. This should then be possible to "restore" the stats.

Zemistr, 2025-08-22:
> Heads up, I noticed today that my door sensor has loosened itself. It was lined up correctly during calibration, but now the sensor always says the door is open even when it's closed which causes the lights to not dim. Keep an eye out for door switches moving, that could possibly be a part of this issue.
> 
> I was wondering why my lights weren't dimming at all, and this is why

Hmmm... my door is opened all the time and the door sensor is turned off.