prusa3d/Prusa-Firmware-Buddy #4772: [FEATURE REQUEST] PT1000 Firmware support

5jvm0u4, 2025-08-20:
### Printer model

Core One

### Describe the feature

I'd like the ability of printing high temp material, with the Core One's chamber, 104NT is the only limiting factor for temperature up to 350C, for a 55C bump up from 295C, we open the world of PET-CF, PPS-CF, HTPA-CF, PPA-CF and more.

It's easy for one to change the hardware, but with the complexity of Prusa's firmware nowadays, even though it's still open sourced, it's already way beyond a normal person's ability to do any modifications on it.

If Prusa could make this available on the firmware, you could also make it an hardware add-on like the ObXidian nozzle to recruit the development cost, and we could have an option, at least for me, I need PPS-CF for my project, but 295C just couldn't do it.

### Expected functionality

Either a toggle switch on the menu, or a auto detection based on the vastly different reading, or a toggle switch plus a suggestion pop-up when reading seems to suggest a PT1000.

3d-gussner, 2025-08-21:
@rykerg9 Can you please publish your changes you have made in the custom firmware on github? This might help to get this feature request to be reviewed/considered.

5jvm0u4, 2025-08-21:
@3d-gussner All material I mentioned above can be printed with 50C chamber temperature or less, PET-CF could even be printed in the open air, not that it would be good, but it can, the only limiting factor is nozzle temperature, they all require over 300C to be able to print.

rykerg9, 2025-08-22:
@3d-gussner I've published the working firmware on my profile, apologies in advance for the lack of organization- I'm getting used to using github for stuff other than just downloading stuff haha.

3d-gussner, 2025-08-22:
@rykerg9 Thanks, I remember how it was for me starting with github.