prusa3d/Prusa-Firmware-Buddy #4773: [BUG] Unfixable Nozzle Cleaning Failure After About 15 Prints

wizzlebippi, 2025-08-21:
### Printer model

Core One

### Firmware version

6.3.4

### Upgrades and modifications

MMU3 Community Mod, Advanced Filtration, Buddy Cam

### Printing from...

USB drive/Prusa Connect

### Describe the bug

After a full factory reset, I can achieve about 15 successful prints on my Core One.  During those prints, the nozzle cleaning performance seems to degrade, eventually failing consistently despite having a clean nozzle and running through Prusa's troubleshooting multiple times.  Then, making no mechanical modifications to the printer, including changing the print sheet or nozzle, cleaning the nozzle, or any other adjustments, I can conduct a hard reset and be back up and running again after having just failed a print due to nozzle cleaning.  Not gcode dependent because I can fail nozzle cleaning and successfully print the same file, just on opposite sides of the hard reset. 

### How to reproduce

1. Use the printer until nozzle cleaning fails consistently. 
2. Attempt reset using button on LCD controller. 
3. Settings --> System --> Factory Reset --> Hard Reset. 
4. Resume Printing.

Might be the only machine that does this. 

### Expected behavior

The printer should work consistently, every time. 

### Files

The printer has not crashed, and I have no files to post.

3d-gussner, 2025-08-21:
@wizzlebippi Thanks for the bug report. Assigned myself to prevent stale-bot doing its thing and created an internal ticket.

wizzlebippi, 2025-08-26:
An update. 

After 4 hours chatting with Prusa support, checking just about everything imaginable mechanically, and me stripping one screw hole in the linear rail support bar, I was directed to perform a more complete hard reset that also cleared the eeprom. While this initially looked a lot like the other hard resets I had performed, there were two notable differences afterward. 

1. In something I hadn't thought about in a while, I can now set my wifi network from the Prusa app via NFC. This had never worked for me before, with my phone never successfully making the handshake with the printer. This had been a problem since I upgraded to the MK4S, and I attributed it to a bad antenna or wire. 
2. My white light bar now dims when the printer is idle.  It never did that before, and I was unaware until last night that it was supposed to. 

While seemingly unrelated, these changes give me hope that an underlying issue has been mitigated.

wizzlebippi, 2025-09-06:
After receiving and swapping the linear rail support bar and printing for a week, it appears that my issue has been solved by the eeprom reset and reinstall of firmware.  I'm guessing that a firmware update long ago caused this issue by corrupting something in eeprom.  

If you've made it this far and have gone through your printer with a fine-toothed comb looking for mechanical issues, don't overtorque things like I did.  Download the attached video, download the latest firmware, take a picture of your stats on your printer (they'll be gone afterward), and fully reset your machine using the instructions from Josef himself.  

[MK4 - EEprom reflashing.zip](https://github.com/user-attachments/files/22189047/MK4.-.EEprom.reflashing.zip)

wizzlebippi, 2025-09-09:
I spoke too soon.  I switched to a PEI sheet from a Satin sheet to print PLA and was greeted with the familiar nozzle cleaning failure.

3d-gussner, 2025-10-03:
@wizzlebippi Thanks for sharing and testing.

wizzlebippi, 2025-10-07:
After some thorough testing over the past few days, aka printing things I hadn't been able to for the past 1.5 months, I'm tentatively declaring this solved.  I would like to add some custom g code to disable the advanced filtration fan during nozzle cleaning and bed leveling to see if that's a long-term solution.