prusa3d/Prusa-Firmware-Buddy #4781: Link becomes unresponsive after a while, can still access via Connect.

kenour, 2025-08-27:
I prefer to use Link for transferring gcode and starting prints, it's faster and my stuff is kept local.

Unfortunately Link seems to die quite frequently, the website just becomes unresponsive and dies.

I restart the machine using Connect, then Link becomes available again, for a while, until it crashes again.

I'm on a 3.5S, with the latest firmware (6.2.4+8909). Any suggestions?

3d-gussner, 2025-08-27:
@kenour Can you please create some logfiles and send these to reports@prusa3d.com. Please share the ticket n umber you get back here, so I can ask for these to check.

1. LCD Settings -> System -> Save Logs To File enable it
2. Keep it running until PrusaLink stops working
3.  LCD Settings -> System -> Save Logs To File disable it to write the log.txt
4. VERY IMPORTANT Do not power off /reset the printer !!!

Other option is to use Putty but connect this time with 57600 baud and send that logfile

Which browser do you use?
What happens when you refresh the browser?

kenour, 2025-08-27:
> [@kenour](https://github.com/kenour) Can you please create some logfiles and send these to [reports@prusa3d.com](mailto:reports@prusa3d.com). Please share the ticket n umber you get back here, so I can ask for these to check.
> 
> 1. LCD Settings -> System -> Save Logs To File enable it
> 2. Keep it running until PrusaLink stops working
> 3. LCD Settings -> System -> Save Logs To File disable it to write the log.txt
> 4. VERY IMPORTANT Do not power off /reset the printer !!!
> 
> Other option is to use Putty but connect this time with 57600 baud and send that logfile
> 
> Which browser do you use? What happens when you refresh the browser?

Logging turned on as requested, I use Chrome/Brave, I get the standard non responsive error as below;

This site can’t be reached
192.168.1.69 took too long to respond.
Try:

Checking the connection
[Checking the proxy and the firewall](chrome-error://chromewebdata/#buttons)
[Running Windows Network Diagnostics](javascript:diagnoseErrors())
ERR_CONNECTION_TIMED_OUT