prusa3d/Prusa-Firmware-Buddy #4788: [FEATURE REQUEST]  Please remove final purge and park from Firmware and place in PrusaSlicer end gcode

pilotltd, 2025-09-02:
### Printer model

Core One

### Describe the feature

I am partially disabled and find it very difficult to reach into back right corner of enclosure to remove the dangling filament residue when a print finishes and have burnt my forearm reaching in there!!  Park position in slicer end code does not work properly! Extruder moves to a modified Y axis position, pauses then moves to the rear right to perform the purge and final park.  By the time it is safe for me to reach in, the filament has hardened and is more difficult to remove without a reheat and manual move to avoid nozzle cleaning failed error. I don't use a camera so this feature would not affect me, for those that do have a camera they can simply leave position as is.

### Expected functionality

Allow final purge and park position at end of print to be user adjustable and not fixed to right rear corner as now.

pilotltd, 2025-09-03:
Thanks my arm is OK now - was very sore for a day or two though! 80c+ on the inside of your forearm is not recommended.,,  

Yes, I did change it  to X242 Y0 F10200 ; park  and also saved it as preset and ran from there.  I started this one off in the forum and someone else tried it also and it didn't work for him either. I don't have an MMU and printer settings are stock with it disabled.
It is specific to the Core One firmware and appears to be set in the firmware. I believe it was at the front originally but Prusa moved it to the back so as not to mask the camera. Pretty pointless having a park position in the custom end code that doesn't park where it's set to and why it is inside firmware doesn't make sense to me?

3d-gussner, 2025-09-03:
@pilotltd Thanks for the update. I will look into it.