prusa3d/Prusa-Firmware-Buddy #4790: [FEATURE REQUEST]

GWdd, 2025-09-02:
### Printer model

Core One

### Describe the feature

When starting prints from the console (physical LCD panel) ...

I select Print from SD, select the file I want to print, press PRINT. Printer starts working, temperatures are climbing, so I go get a cup of coffee.

Minutes later (30 minutes, hour, etc.) I walk by the printer and see it is waiting for me to press the ARROW button, and the process repeats. 

Seriously?  I thought selecting the file and pressing PRINT told the printer I wanted to print the file. Why do I need to re-affirm I really really really want to print the file?

Give me a switch for the ARE YOU SURE ? function, please. Or, just deletete it, as it serves no function I can understand.  If I accidentally start a print, I can press CANCEL. 

### Expected functionality

I start a print, the printer prints it.

3d-gussner, 2025-09-03:
@GWdd Can you please explain bit more what happens? Normally the printer will start printing when you select print.
Which message have you seen that you needed to confirm?

3d-gussner, 2025-09-04:
When you select a print from USB Stick it will show you the preview (picture you see) and you need to press print again.
As users may accidentally select the wrong file in first step they can go back. Especially with the touch screen it is possible to "miss" the correct filename. Also the preview is an visual indicator what will be printed and again enables the user to see if that it is what they want to print.

This will not be changed as the existing process is working well. Starting the print immediately probably would create more frustrations and issues.