prusa3d/Prusa-Firmware-Buddy #4791: [BUG] MMU3 filament change halting a print

CappyT, 2025-09-02:
### Printer model

Core One

### Firmware version

6.3.5-alpha+10514

### Upgrades and modifications

MMU3 Enclosed kit

### Printing from...

Prusa connect (usb inserted) 

### Describe the bug

The printer stucks in the loading filament screen (like when you load filaments on mmu) and doesn't do anything else, failing the current print. 

### How to reproduce

- Run a print
- Make the mmu3 finish the filament mid print
- Make it so that the mmu cannot unload the filament properly, causing an error on the screen (i think it tries 3 times before showing)
- Now "fix" the issue and let filament unload
- Load new filament
- The loading filament screen doesn't go away. 

### Expected behavior

Filament loads and proceeds to resume printing 

### Files

N/A

markaabo, 2025-09-21:
> What is likely the same bug is still present in 6.4.0 alpha.
> 
> MK3.9S with MMU3 Firmware: 6.4.0-alpha+11774 MMU Firmware: 3.0.3
> 
> I find the printer in the parked position with "Reloading Filament" at 100 % and "Parking" showing on the screen. The filament is not unloaded from the hotend and the printer does not react to any button presses except reset.
> 
> I have not been monitoring the printer, but I assume this happens when a filament swap fails and it is parking waiting for user resolution.

I also have this issue, for me it just gets stuck at 100% without trying to unload the filament, i have not heard the mmu3 make the "pulling" sound when doing it (aka the stepper), so it seems to happen before it actually starts swaping, for me it mostly happens on my number 5 filament but that is always my support filament which is used the most so could be just by chance. but just wanted to add it as a note.

drGarbinsky, 2025-12-10:
I too have this issue on a mk3.9. 6.4.0 final release

m42e, 2025-12-26:
I am unsure if this helps but i managed to activate syslog before the "crash" happened:

```json
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"MarlinServer","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782541665866824,"message":"E-stall detection off"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782547604758598,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782549605394216,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782551604702348,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782554782882995,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782556783974522,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782558782241744,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782560782242083,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782562782213162,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782564957617418,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"MMU2","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782566215504622,"message":"Unloading to FINDA"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782566957948092,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782568957715856,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"MMU2","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782569285068982,"message":"Retract from FINDA"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782570957799270,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782572958733149,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782574324458711,"message":"RSTCommTimeout"}
[... cut ...]
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782953059442881,"message":"busy: processing"}
{"level":"info","log_processor":"true","app_name":"buddy","client":"192.168.xxx.xxx","hostname":"MAC-ADDRESS-XXXXX","priority":"14","proc_id":"Marlin","msg_id":"-","severity":"Informational","facility":"1","structured_data":"-","tls_peer":"","version":"1","time":1766782955059371634,"message":"busy: processing"}
```

cgspeck, 2026-01-03:
This hit me today, multiple times.
The first on hour 4 of a 7 hour print. 
The second when I attempted to start a print. 
This bug has rendered my MK4S + MMU3 unreliable and unusable.
On 6.4.0 final (6.4.0+11974)

m42e, 2026-01-12:
Ok, i hope I got it right, but here is what happens according to my analysis:

## 1. Overview
The message **"Reloading filament"** appears on the screen when the printer enters the **Filament Stuck** recovery mode. This is distinct from a standard filament runout or a user-initiated filament change (`M600`).

## 2. Trigger Mechanism
The "Reloading filament" state is triggered automatically by the **Extruder Motor Stall Detection** system. This feature is enabled on printers equipped with a Load Cell.

### Sequence of Events:
1.  **Stall Detection**: The `EMotorStallDetector` inside `Marlin.cpp` detects a stall on the extruder motor while the axis is moving and pushing filament.
2.  **G-Code Injection**: Upon validating the stall, the firmware injects the internal G-Code **`M1601`**.
    *   *Note*: On CoreOne printers, this is currently suppressed/delayed.
3.  **M1601 Execution**: `PrusaGcodeSuite::M1601` calls the filament change executor `M600_execute`.
    *   It passes `is_filament_stuck = true`.
4.  **Pause FSM Setup**: `M600_execute` calls `Pause::filament_change`.
    *   Ideally, `load_type` is set to `LoadType::filament_stuck` (instead of `filament_change`).

### Code Execution Path
Here is the detailed step-by-step execution path from the `M600` call to the MMU hang:

1.  **M1601 / M600 Call**:
    *   **File**: `src/marlin_stubs/pause/M600.cpp`, Line 264
    *   **Code**: `Pause::Instance().filament_change(settings, is_filament_stuck);`
    *   **Action**: Calls the `filament_change` method on the singleton `Pause` instance.

2.  **Initialization**:
    *   **File**: `src/marlin_stubs/pause/pause.cpp`, `Pause::filament_change`
    *   **Logic**:
        *   Sets `load_type` to `LoadType::filament_stuck` (if triggered by M1601) or `LoadType::filament_change`.
        *   Calls `invoke_loop()` to start the state machine.

3.  **FSM Loop Start**:
    *   **File**: `src/marlin_stubs/pause/pause.cpp`, `Pause::invoke_loop`
    *   **Logic**:
        *   Calls `set(LoadState::start)` to initialize the state.
        *   Enters a `while` loop that executes the function pointer corresponding to the current state from the `state_handlers` array.

4.  **State: `LoadState::start`**:
    *   **Handler**: `Pause::start_process`
    *   **Logic**:
        *   Switches on `load_type`.
        *   For `LoadType::filament_stuck` (or `filament_change`), it calls `set(LoadState::unload_start)`.

5.  **State: `LoadState::unload_start`**:
    *   **Handler**: `Pause::unload_start_process`
    *   **Logic**:
        *   Checks `if (FSensors_instance().HasMMU())`.
        *   Since an MMU is connected, it enters the `if` block.
        *   Calls `set(LoadState::mmu_unload_start)`.

6.  **State: `LoadState::mmu_unload_start`**:
    *   **Handler**: `Pause::mmu_unload_start_process`
    *   **Action**: This is the function where the logic gap exists.

The issue is that `Pause::mmu_unload_start_process` does **NOT** handle `LoadType::filament_stuck`

```cpp
void Pause::mmu_unload_start_process([[maybe_unused]] Response response) {
    if (load_type == LoadType::unload) {
        // standard unload logic
    } else if (load_type == LoadType::filament_change) {
        // standard M600 logic
    }
    // implicit else: do nothing
    return;
}
```

Prusa-Support, 2026-02-28:
Thanks for reporting.
Would it make any difference if the feature `stuck filament detection` is disabled or the firmware is downgraded to 6.2.6 (MK4)?
If so, we may be looking at a known bug, and the bugfix seems to be currently in progress.

Michele Moramarco
Prusa Research