prusa3d/Prusa-Firmware-Buddy #4792: [FEATURE REQUEST] Better temperature logic during filament change (High temp to low temp)

geco0on, 2025-09-05:
### Printer model

Core One

### Describe the feature

I have experienced this more times than I can count. I print all kinds of materials. This is why I constantly run into this problem.
When swapping from high temperature material, to low temp material, when pressing ,,change filament'' option on printer menu, My latest example, PC Blent swap to PLA, It always preheats to PLA temperature, never to PC, and when it tries to extrude and purge old filament, gears start grinding and it clicks a lot in nextruder, I find chunks of plastic afterwards in it, sometimes it almost doesn't extrude.
My suggestion is to change the logic or something, and preheat it enough to melt high temp plastic (in this example PC), and then lower temperature, to not burn the PLA.
What are opinions on this?

m-cas, 2025-09-09:
Interesting, I usually "unload filament" instead of "change filament" and the old filament gets completely ejected form the extruder. Then I simply push in the new one and select the filament type from the menu that appears when the sensors sees the new material. This has worked for me so far rather well even from high temp to low temp. It would appear that "change filament" has a different logic than just "unload + load"...

LordMelkor, 2025-09-09:
@m-cas I think you've misunderstood the issue. If you used materials with a more extreme difference you'd be more likely to notice. Even with unload instead of change, when you do the subsequent load, it heats the nozzle to the set temp of the new material (not the old).

When you extract the filament with unload, the filament comes out, but there will always be some filament left in the meltzone of the nozzle. That is why a purge is needed. The high temp material to low temp material becomes problematic because the remnant of the high temp material may not get completely purged with insufficient temperature.

Also please note that github issues are not the same as a reddit discussion forums, and are not meant for open-ended discussion.

m-cas, 2025-09-09:
@LordMelkor what for you is open-ended discussion is for me "framing the problem"; please refrain from the teaching attitude.