prusa3d/Prusa-Firmware-Buddy #4799: [BUG] Prusa XL gets stuck in tool assignment screen after Homing Calibration is recommended

eqdeng, 2025-09-10:
### Printer model

XL 5T

### Firmware version

6.4.0-alpha

### Upgrades and modifications

N/A

### Printing from...

Prusa Connect

### Describe the bug

In the tool mapping screen, I attempted to map slicer tool 1 to printer tool 3 and carry out a filament change. The tool was picked, and the new (as of today) homing procedure ran. I got a message that it was recommended to run the Homing Calibration. (Options are Abort or Ignore.) I selected Abort, and was sent back to the mapping screen. Pressing the knob button on Back does nothing. I was not able to exit the print job without resetting the printer.

braw, 2025-09-13:
@3d-gussner Yes, I ran the X cal, Y cal, and Homing cal, but was still experiencing tool head crashes.

braw, 2025-09-13:
@3d-gussner I didn't run all calibrations again. When I flashed 6.4.0-alpha, the fan test and homing calibration were marked as not being run (dash instead of green check or red x), so I ran the Fan cal, X and Y cal just in case, and Homing cal.

braw, 2025-11-21:
There is a clunk, as if it hits something it didn't expect like a crash. I took a few more videos. maybe these are normal behaviors, but I don't recall so many rough clunks like I am seeing now.

Here is one where I already homed it, picked and parked a tool, moved the gantry to roughly the middle in X and Y the move axis in the menu. Then selected auto home. [Here is the behavior](https://youtu.be/odnXz5aeG9U).

I also added some text to another attempt to auto home to highlight the exact moment I was originally talking about. [Here is that video](https://youtu.be/9SoobwbE8XU).

Kleiner-Iltis, 2025-11-25:
Same for me, can confirm that happens to my machines also on 6.4.0 RC2 multiple times. Also making sometimes loud noise when homing. Sound like the stepper drivers are doing something wrong. I executed also the recommended recalibration. Had to go back to 6.2.6, after several crashes.

braw, 2025-12-02:
@Prusa-Support If you watch the [video here](https://youtu.be/9SoobwbE8XU?t=44), the tool changer clunks or crashes into the left side when auto homing. This is something that started with v6.4.0-alpha, and continues into v6.4.0 final. If I roll my XL back to v6.2.6, this behavior is not present as [seen here](https://youtu.be/hmcloZGEUqY?t=10). 

My belts are perfectly tuned, all calibrations have been re-run, IS cal and phase stepping re-run. The only variable that changed is the firmware version.

If the new clunk/crash is intentional and by design, that is surprising.

Prusa-Support, 2025-12-03:
. 6.4.0-alpha was problematic ❌ -> 6.4.0 solved it ✔️ 
. semi-crash ❓ -> no panic 🙂 -> please follow #4972 🙏