prusa3d/Prusa-Firmware-Buddy #4811: [ENHANCEMENT] Prusa Link API should provide more information in

edasque, 2025-09-13:
### Printer model

XL Multi-tool

### Describe the enhancement

Whether on ```/api/v1/status``` or another endpoint, the printer knows what material is loaded in each of the tools but it is not reported by the Prusa Link (or Prusa Connect) API. I always lose track of materials and would like to build an application so I can track the current color and material loaded onto the printer(s) tools so I don't have to physically check.

The status API (or a dedicated endpoint) should report the status of each tool included material loaded. This would be useful for PrusaSlicer as well and could reveal discrepancies at upload time between the design and the reality of the physical printer. 

### Expected functionality

```/api/v1/status``` or a dedicated endpoint reports on each tool status including material loaded if any

edasque, 2026-02-15:
@3d-gussner you think it’s something you could tackle?

boecko, 2026-03-04:
I've made a similar request #5148

closed 2026-10-05 by github-actions[bot]