prusa3d/Prusa-Firmware-Buddy #4816: [BUG]

tfhh, 2025-09-14:
### Printer model

CoreOne

### Firmware version

6.4.0 alpha

### Upgrades and modifications

BSOD when print with large height is finished

### Printing from...

PrusaConnect

### Describe the bug

When I print objects with large height (> 200mm) the print itself will be done fine until the end, but when the extruder is moved to after-print park position and the bed is lifted up, the printe resets and showing a BSOD. Have had this three times with three different models

### How to reproduce

Just print any high model (>200mm).

### Expected behavior

[image]

[image]

### Files

Crash dumps are sent to the email address mentioned.

DRracer, 2025-09-16:
Can you please post the G-code you printed? At least zip it and send to the same email address like you did with the dumps.

What I see from the dump: a move is being planned to X=242, Y=211, Z=39 (!) while the max printed Z was recorded at 245 - no wonder the printer objects (it would be nicer if if didn't crash, but that's something else 😉  )

DRracer, 2025-09-18:
@tfhh great, I've been able to reproduce the issue + verify the fix.
Btw.: where did you get the line `M84 Z X Y E` in the end gcode?

tfhh, 2025-09-18:
@DRracer Sounds good! I´ve added the "Z" to the disable motors command to have the printer quiet when it is ready. The hissing sound of three Z motors when idle is annoying :-)

closed 2026-02-14 by Prusa-Support