prusa3d/Prusa-Firmware-Buddy #4819: [BUG] Severe layer shift on crash recovery and re-homing [XL 6.4.0-alpha1]

davispw, 2025-09-14:
### Printer model

XL

### Firmware version

6.4.0-alpha1

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

With 6.4.0-alpha1 I'm seeing frequent crashes on tool change - there are other issues covering that.  This issue is about severe layer shift upon recovery and re-homing.

### How to reproduce

1. Print the attached project - bed 3 (the "planter box + dirt")
2. Crashed on the first tool change between layers 4 and 5 (when returning tool 1, before picking up tool 2)
3. On the XL, recover by manually caging tool 1 and clicking continue
4. Observe automatic re-homing, after which it picks up tool 2 and resumes

Result: tool 2 is **shifted exactly 5mm in the +Y direction.**

[image]
[image]

_I've reproduced the same issue 3 times with different prints with 6.4.0-alpha1._  I've successfully recovered crashes in the past with 6.2.x, so this seems to be a regression with the new homing logic.

### Expected behavior

It should recover after re-homing correctly.

(Also it shouldn't crash - separate issue)

### Files

[carrot-garden.3mf.zip](https://github.com/user-attachments/files/22322104/carrot-garden.3mf.zip)

[carrot-garden_0.4n_0.2mm_PLA,PLA_XLIS_5h3m.bgcode.zip](https://github.com/user-attachments/files/22322651/carrot-garden_0.4n_0.2mm_PLA.PLA_XLIS_5h3m.bgcode.zip)

3d-gussner, 2025-09-15:
@davispw Thanks for testing 6.4.0-alpha on the XL and the feedback.

Jeggo65, 2025-09-24:
I had several same issues on my XL. But it seems to be the slicer and not the firmware. If the model sliced with Prusa slicer (latest version) the problems occur. If the same model is sliced with Orca slicer, the print is OK.

3d-gussner, 2025-09-25:
@Jeggo65 Would it be possible to share the gcode files from PrusaSlicer and OrcaSlicer?
Or do you see obvious setting differences between the two slicers? Use Arc fitting `G2/G3` commands or avoid perimeters?

Jeggo65, 2025-09-27:
Sorry I can not share the gcode, as I delete failed models/prints/gcode after printing to avoid a second try.
I tried to setup the same configuration in OrcaSlicer as in PrusaSlicer. Unfortunatley you have to do this by hand. So I tired to use the same setting in both slicers. PrusaSlicer seems to be more aggressive with acceleration during moves to the next print position. In same cases the print head collides with the model and causes the layer shift. In most cases the model stay in place.
It makes a awful noise. The crash detection is on the lowest setting in the firmware.

closed 2026-05-12 by github-actions[bot]