prusa3d/Prusa-Firmware-Buddy #4820: [BUG] Failed Tool Change

whjoyjr, 2025-09-14:
### Printer model

XL 5 tool

### Firmware version

6.4.0-alpha

### Upgrades and modifications

none

### Printing from...

Prusa Connect

### Describe the bug

Updated to 6.4.0 via Prusa Connect.  Prior to update tool-changing was fine and worked correctly a few days prior.  Machine was consistently failing performing a tool change to Tool 5.  Performed a fulklk re-calibration including dock calibration and tool-head off-set.  Started failing tool-change on Tool 3.  Reverted to 6.2.6, no change to file, and no tool-change failures.  PrusaSlicer 2.9.3. 

### How to reproduce

Any file requiring tool-changing will trigger the issue

### Expected behavior

On tool change, the capture bars to not correctly capture the toolhead.

### Files

[MegaAmpharos_0.4n_0.2mm_PETG,PLA_XLIS_2h37m.bgcode.zip](https://github.com/user-attachments/files/22322984/MegaAmpharos_0.4n_0.2mm_PETG.PLA_XLIS_2h37m.bgcode.zip)

Paad74, 2025-09-15:
I have the same problem. I have the failure on all five tools. I have done calibrations and belt checks, but the problem was still there. I went back to version 6.2.6. Here the tool changes are ok again.

Essbante, 2025-09-15:
Since the issues is already reported, I'm just confirming tool change failing problem (parking fails) introduced by the v6.4.0-alpha1 firmware.

daavery, 2025-09-16:
One more x/y step loss leading to tool change failure report

s3anami, 2025-09-17:
One more report

Dres76, 2025-09-17:
Same problem here. With one addition: 
I seem to hear something that sounds like random "hickups" of the xy-system during homing, bed probing and toolhead check before the print. 
It sounds like it was missing steps... maybe?

kayblasto55, 2025-09-17:
I got the same time happen twice on one print, put it back manually each time and then the was fine.

https://github.com/user-attachments/assets/516b1fbc-b8a5-482b-8105-57b2e1d5cb86

3d-gussner, 2025-09-25:
@mudddler13 Thanks for testing and reporting the alpha, very appreciated. Understandable that you went back to the stable firmware.

Thanks to everyone commenting here. Developers look into it and 🤞 it will be solved in next final release.

HEEROCKSS, 2025-11-15:
Same problem. Had my XL 5th for a year now and not a single tool change crash. Updated to 6.4.0 Alpha and now facing constant crashes, even when printing single color and the printer tries to park the tool. I took some time watching it and it feels like its parking way faster than with the old firmware, causing it to crash.