prusa3d/Prusa-Firmware-Buddy #4823: Accessibility issue with toggle switch colors – improve ON/OFF contrast

rbarbrow, 2025-09-14:
### Printer model

MK4 or higher

### Describe the enhancement

Currently, the ON/OFF toggle switches in the printer UI shift only slightly in color (white → light gray). This subtle change is difficult to distinguish, especially for users with color vision deficiencies.

Accessibility Issue:

White vs. light gray provides very little contrast

Users with color blindness or reduced vision may not be able to easily determine the toggle state

Suggested Improvements:

Use distinct, high-contrast colors for ON and OFF states

Examples:

- ON: Green (or another high-visibility color)- 
- OFF: Black or dark gray

Alternative colorblind-safe combinations could include:

- ON: Blue, OFF: Dark Gray- 
- ON: Orange, OFF: Black

Ensure contrast ratio ≥ 4.5:1 to meet accessibility guidelines

Benefit:
Improves usability and accessibility for all users, particularly those with visual impairments.

3d-gussner, 2025-09-15:
@rbarbrow Thanks for the idea. Can you please provide some more information about the accessibility guidelines.
Which toggle to you mean on which printer?

closed 2025-09-15 by rbarbrow

rbarbrow, 2025-09-16:
[image]

The UI on the mk4s is also inconsistent with the corresponding UI in PrusaSlicer 2.9.3

rbarbrow, 2025-09-16:
here is the UI on the mk4s

[image]

rbarbrow, 2026-02-15:
Any word on this?