prusa3d/Prusa-Firmware-Buddy #4826: [BUG] The loadcell is once again too sensitive, can't Z home again at start of print!!

CpuNut, 2025-09-15:
### Printer model

Core One

### Firmware version

6.4.0-alpha+11774

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

I seem to have the CoreOne that has the over sensitive loadcell.  What I had to fix in the early code, and then was fixed by Prusa is now broken again.

Auto Home works perfectly, but at the beginning of a print it goes into the crazy Z hunting again, never properly finding the plate.

Can you please add back in as a permanent feature, the loadcell sensitivity setting!

My printer is 100% unusable with the consolidated Alpha firmware.

### How to reproduce

Try a print.

### Expected behavior

As with the last production release, desensitize the loadcell during initial homing and plate leveling.

Better yet, please add back in the Loadcell sensitivity setting in the code.  I don't want to have to compile every code release myself to get the engineering menus back to make the printer work.

antonio-c78, 2025-10-21:
I’m experiencing exactly the same issue on my new CoreOne, which I purchased fully assembled just a few days ago. The load cell problem makes the printer unusable in about 8 out of 10 attempts — it fails to home the Z axis correctly and often cannot perform the nozzle cleaning routine.

I’ve already contacted Prusa Support, but I’m concerned they won’t be able to help, as I’ve found many other reports describing the same issue, yet no real solution so far.

Alienbliep, 2026-03-29:
Same problem here too. On new CORE One L when the bed homes it stops around 20 times in it's up travel. Takes so much time. Swapped the loadcell with antoher from core one and same issue. Loosened the bolts to the point where it think it was too lose. Did not help.