prusa3d/Prusa-Firmware-Buddy #4827: [BUG] Thermal runaway

geco0on, 2025-09-15:
### Printer model

Core one

### Firmware version

6.4.0 Alpha

### Upgrades and modifications

None

### Printing from...

USB and Prusa Connect

### Describe the bug

Experiencing Thermal runaway more often than ever. (Doesn't state which sensor)
It leads by Printer restarting and afterwards popping up Red screen...
It didn't happen during printing for now. Usually during preheat to PC temperatures, and Preheating to change filament. 
Will observe more and give updates in the comments!
(It Didn't happen before alpha firmwares, for me)

3d-gussner, 2025-09-16:
@geco0on Thanks for testing 6.4.0-alpha.
Can you please connect your printer via USB to a computer and use [putty](https://help.prusa3d.com/article/crash-dump_364959) or another serial monitor to get some log files?
Before you start to preheat please send
- `M503` ; dump settings
- `M155 S1` ; auto report the temperatures.

geco0on, 2025-09-16:
It didn't crash on the first next preheat.
I will try to do that every time I preheat, and send You log when it happens.
Please respond if this file would be sufficient next time, so I don't have to hussle about bringing Windows PC for now, since I have a macOS laptop. I used app called ,,SerialTools'' it should do same business as Putty. 

[First preheat after crash.rtf](https://github.com/user-attachments/files/22362279/First.preheat.after.crash.rtf)

3d-gussner, 2025-09-16:
@geco0on If you have again issues especially with high temp filaments you can try this

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4490#issuecomment-3157470376

geco0on, 2025-09-16:
Okay, I menaged to catch it when it crashed. First file is when It crashed. I ran M155 S1 beforehand.
2nd file is next startup with M503 followed by M155 S1.
2nd print started successfully.

[Preheat after 2nd crash.rtf](https://github.com/user-attachments/files/22367969/Preheat.after.2nd.crash.rtf)
[Log while crashing.rtf](https://github.com/user-attachments/files/22367968/Log.while.crashing.rtf)