prusa3d/Prusa-Firmware-Buddy #4830: [FEATURE REQUEST] Reimplamantation of V6 Nozzle Change Routine

4rdful, 2025-09-16:
### Printer model

MK4, MK4S, MK3.5, MK3.5S, Mini+, XL, CoreOne

### Describe the feature

The nextruder nozzles are quite fancy, sure.
However, being a user of a MK3.5/ MK3.5S it is very frustrating changing V6 nozzles, especially knowing there is a feature to automate this procedure for the old firmware of the MK3S+! Would be gorguoes to have it reimplamanteted for the buddy firmware!

Feature is also interesting for all the other Prusa printers (MK4/S, XL, C1, Mini+), when making use of the **V6 nozzle adapter**. Yes, new costumers are likely to use the nextruder nozzles. There are surely a bunch of people, though, using the adapter and making it worth to reimplement this feature! ;)

### Expected functionality

The feature should be accessed by a context menu, maybe via "Control/Nozzle (V6) change".

In particular the feature should follow the old routine:

1.  Setting the extruder as high and central as possible,
2. heating the thing to the corresponding temperature and 
3. giving a short wizard to click through while changing the nozzle.

Optionally, it could show a QR-code for further support, that links to the Prusa website where the procedure is well explained ([for MK3.5/S](https://help.prusa3d.com/article/changing-or-replacing-the-nozzle-mk2-5-s-mk3-s-mk3-5-s_2069) + [for V6 adapter](https://help.prusa3d.com/article/installing-or-changing-the-nextruder-v6-nozzle-adapter_823792)). Needless to say, bed temperature should remain off.

4rdful, 2025-09-25:
@3d-gussner Cool ^^
Do not know exactly, how developing works internally. Whether you further consider the feature and its possible options or just implement it the way I suggested. If so, it could also be asked which nozzle was installed directly after the procedure is done (e. g. by counting the holes in the nozzle and entering by the printer's scrolling knob or else via touchscreen)

3d-gussner, 2025-09-26:
Feature requests are reviewed and the suggested ideas are considered.
There have been some that more or less have been implemented as suggested others differently to be more usable to the wide range of users (beginner, normal, advanced or pro).

I can't tell when and if it will be implemented.

For the time being I would suggest a simple Nozzle_change.gcode
```
G28 ; Home
G1 X125 Z200 F1000 ; move extruder
M109 S280 ; Preheat to 280°C
M84 XY ; Disable XY motors
M117 Change Nozzle
M601 ; Pause until Nozzle has been changed
M104 S0 ; turn off temperature
M117 Change Nozzle size in Menu
M601 ; Pause to ensure message being shown on LCD
M83
```

4rdful, 2025-12-12:
Comment as a try to keep issue opened, Dec. 2025.

4rdful, 2026-02-16:
Comment as to keep issue opened, Feb. 2026.

3d-gussner, 2026-02-17:
@4rdful Assigned this issue to myself to prevent stale-bot from closing it. Frankly I can't give you an update if and when it may be implemented.