prusa3d/Prusa-Firmware-Buddy #4837: [BUG] First Layer Calibration MK3.5S / Firmware 6.4.0-alpha+11774

juckendesAuge, 2025-09-16:
### Printer model

MK3.5S

### Firmware version

6.4.0-alpha+11774

### Upgrades and modifications

Revo6 Hotend

### Printing from...

Only First Layer Calibration.

### Describe the bug

First Layer Calibration.
It is not possible to change the Z height with the rotary wheel. It is grayed out.
Even after the purge line.

I don't know if it has anything to do with the fact that I clicked Reset for some profiles under Settings, Hardware, Steelsheets.
The labeling still looks normal there. The custom profiles for the steel sheets have not disappeared.

However, when you click on Control, Sheet Profile, some of them are marked with a question mark.

[image]

[image]

[image]

[image]

schrodlm, 2025-09-19:
I have worked on this issue this week and the solution proved to be a little bit tricky:D But it is fixed and will be part of 6.4 release.

3d-gussner, 2025-09-26:
FYI should be fixed in final release.

3d-gussner, 2025-11-19:
@ablaman @ablaman @schrodlm Please try firmware [6.4.0-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-RC2) and report back.

juckendesAuge, 2025-11-21:
Sorry, I can't help right now.
Unfortunately, I have to prepare for Christmas...

closed 2026-02-14 by Prusa-Support